gumyr/build123d #364: Add build123d method for listing fonts available to OCP/OCCT

jdegenstein, 2023-11-03:
Rough prototype:

```py
mgr = Font_FontMgr.GetInstance_s()
abc = mgr.GetAvailableFonts()
for i in abc:
    print(i.FontName().ToCString())
```
Returns:
```
Arial
Arial Black
Bahnschrift
Bahnschrift Light
Bahnschrift SemiLight
...
```

gumyr, 2023-11-04:
Thanks @jdegenstein, this would be good to have.  It's somewhat like `import_svg_as_buildline_code` more of a tool then a direct construct CAD capability.  I guess if it's created as a function like `list_available_fonts()` then it doesn't really matter how it's classified.

rawwerks, 2024-02-07:
+1 for this feature, would make my life so much easier, especially since i'm hosting b123d (so the user won't have any way of directly checking the server for available fonts).

PR #1051 (2025-07-30): Add method to list available system font names and FontStyles
 I've been sitting on this for a while and #364 was referenced by @jdegenstein on discord, so figured I should share it.

* Adds public `available_fonts()`
* Returns list of objects sorted by name (which `GetAvailableFonts()` doesn't do)
* Didn't find a suitable module for this so made a new root module `utils.py` for non-geometry related helper functions
* Resolves #364

I'm open to changing pretty much anything about this including output format and location.

Best viewed with pprint:
```py
from pprint import pprint
pprint(availab [... cut]