gumyr/build123d #368: PolarLocations does not span angular_range

anderson-pa, 2023-11-06:
Positions returned from PolarLocations do not include the end of the range. This yields the expected behavior for the default angular_range=360 (N evenly distributed points around a circle) but not for smaller ranges. For example, count=3 and angular_range=90 returns angles of (0, 30, 60) instead of (0, 45, 90).

The function is simply calculating too many gaps for the latter case:
https://github.com/gumyr/build123d/blob/630c7c65b62c1a0a83529a4c4ff00144c4343a90/src/build123d/build_common.py#L908

Existing test case:
https://github.com/gumyr/build123d/blob/630c7c65b62c1a0a83529a4c4ff00144c4343a90/tests/test_build_common.py#L151-L159

jdegenstein, 2023-11-06:
This is a surprisingly tough issue to handle, I wonder if this solution is superior to what is currently in place:
```py
class PolarLocations ...
...
        if angular_range % 360 == 0:
            angle_step = angular_range / count
        elif count > 1:
            angle_step = angular_range / (count - 1)
        else: #handle ==1 case
            angle_step = angular_range
```
Maybe a tolerance needs to be applied to the above modulo comparison?

The above gives the following behavior with this code:
```py
num, start, range = 6, 60, 360
locs = PolarLocations(1, num, start,range).local_locations
zz = [loc.position for loc in locs]
show(zz,PolarLine((0,0),1,start+range))
```
[image]

anderson-pa, 2023-11-06:
Huh, I hadn't considered the count=1 case but that does make sense. Well at least for me, your solution fits my intuition of the expected behavior.

Is there a use case for range > 360? For example, could these locs can be used on a helical path?
* If no, then maybe sufficient to assume 0 < range <= 360 and only omit the last point if range==360
* If yes, then desired behavior might be to keep the last point when range==N*360 (still thinking of the helical path in this case). Problematic for the default case, where most of the time user would expect the point to be omitted but if they want to do something unusual, would need a way to override.

snoyer, 2023-11-07:
> Maybe a tolerance needs to be applied to the above modulo comparison?

What about changing `angular_range: float = 360.0` to `angular_range: Optional[float] = None` to distinguish the default "full circle" case?

Obviously it would be nice to start with the _what_ before the _how_ and clarify the expected behavior to before debating implementation.

gumyr, 2023-11-07:
The idea behind this was to work like Python's `range` (originally there was a `stop` not a `range` parameter IIRC):
```py
>>> print([i for i in range(0,90,30)])
[0, 30, 60]
```
however, I agree that this might not be natural.

The `numpy` `linspace`, `logspace` and `geomspace` functions all have an `endpoint: bool = True` parameter and works like this:
```py
>>> from numpy import linspace
>>> print([i for i in linspace(start=0,stop=90,num=3,endpoint=True)])
[0.0, 45.0, 90.0]
>>> print([i for i in linspace(start=0,stop=90,num=3,endpoint=False)])
[0.0, 30.0, 60.0]
```
so how about I add an `endpoint` parameter to `PolarLocations` so it works like the numpy functions?  With a default `range` of `360` the default value of `endpoint` would be `False`.  What do you think of this proposal?

gumyr, 2023-11-08:
Added the `endpoint` parameter with commit cdd00771f50bebcf52d922561d2eb761ed0064d8