gumyr/build123d #369: 2D chamfering very slow on Apple Silicon

MatthiasJ1, 2023-11-08:
The below code will run 2 orders of magnitude slower on an Apple M1 vs x86 hardware. 3D chamfer operations are equally fast to x86.

```py
from build123d import *
import time

start = time.time()

with BuildSketch():
    Rectangle(10,10)
    with BuildSketch(mode=Mode.SUBTRACT) as s:
        Rectangle(8, 8)
        chamfer(s.vertices(), 1)

print("Time:", time.time()-start)
```

jdegenstein, 2023-11-08:
It is interesting how different the approach for `TopTools_IndexedDataMapOfShapeListOfShape` is for `chamfer()` vs `chamfer_2d`. The latter uses `_entities_from()` whereas the former has built-in code. I assume they produce similar results??

https://github.com/gumyr/build123d/blob/040292131705dda5a65dd37369573d4827fb9bfc/src/build123d/topology.py#L2107-L2126

https://github.com/gumyr/build123d/blob/040292131705dda5a65dd37369573d4827fb9bfc/src/build123d/topology.py#L5527

https://github.com/gumyr/build123d/blob/040292131705dda5a65dd37369573d4827fb9bfc/src/build123d/topology.py#L1095

gumyr, 2023-11-10:
Unfortunately, I don't have access to a Apple silicon based Mac and therefore can't investigate this issue at this time. Maybe someone with the appropriate hardware could attempt to track down the problem?

anb, 2024-01-10:
I ran it on an M3 and I also get much slower times with 2D chamfer :

2D 1.4228092079865746
3D 0.008684917003847659

Time is spent in _entities_from in topology.py as mentioned above, specifically res.FindFromIndex in the list comprehension which takes 300ms each time. 

Let me know if I can try something.

jdegenstein, 2024-01-10:
@anb From an ipython console with `line_profiler` installed could you run the following:
```py
from build123d import *
from build123d import Shape
%load_ext line_profiler
%lprun -f Shape._entities_from -f Face.chamfer_2d chamfer(Rectangle(10,10).vertices(), 1)
```
on my x86 machine I get the following (FindKey seems to be one bottleneck):
```py
Timer unit: 1e-07 s
Total time: 0.0048188 s
File: c:\Users\someuser\.conda\envs\ocp_vscode\Lib\site-packages\build123d\topology.py
Function: _entities_from at line 2104

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
  2104                                               def _entities_from(
  2105                                                   self, child_type: Shapes, parent_type: Shapes
  2106                                               ) -> Dict[Shape, list[Shape]]:
  2107         1         90.0     90.0      0.2          res = TopTools_IndexedDataMapOfShapeListOfShape()
  2108                                           
  2109         2        131.0     65.5      0.3          TopExp.MapShapesAndAncestors_s(
  2110         1          3.0      3.0      0.0              self.wrapped,
  2111         1          6.0      6.0      0.0              inverse_shape_LUT[child_type],
  2112         1          3.0      3.0      0.0              inverse_shape_LUT[parent_type],
  2113         1          2.0      2.0      0.0              res,
  2114                                                   )
  2115                                           
  2116         1          3.0      3.0      0.0          out: Dict[Shape, list[Shape]] = {}
  2117         5         97.0     19.4      0.2          for i in range(1, res.Extent() + 1):
  2118         8      47131.0   5891.4     97.8              out[Shape.cast(res.FindKey(i))] = [
  2119         4        720.0    180.0      1.5                  Shape.cast(el) for el in res.FindFromIndex(i)
  2120                                                       ]
  2121                                           
  2122         1          2.0      2.0      0.0          return out

Total time: 0.0068371 s
File: c:\Users\someuser\.conda\envs\ocp_vscode\Lib\site-packages\build123d\topology.py
Function: chamfer_2d at line 5571

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
  5571                                               def chamfer_2d(
...
  5595                                           
  5596         1       2529.0   2529.0      3.7          chamfer_builder = BRepFilletAPI_MakeFillet2d(self.wrapped)
  5597         1      48444.0  48444.0     70.9          edge_map = self._entities_from(Vertex.__name__, Edge.__name__)
  5598                                           
  5599         5         51.0     10.2      0.1          for vertex in vertices:
  5600         4        544.0    136.0      0.8              edges = edge_map[vertex]
  5601         4         48.0     12.0      0.1              if len(edges) < 2:
  5602                                                           raise ValueError("Cannot chamfer at this location")
  5603                                           
  5604         4         17.0      4.2      0.0              if edge:
  5605                                                           if edge not in edges:
  5606                                                               raise ValueError("One or more vertices are not part of edge")
  5607                                           
  5608                                                           edge1 = edge
  5609                                                           edge2 = [x for x in edges if x != edge][0]
  5610                                           
  5611                                                       else:
  5612         4         38.0      9.5      0.1                  edge1, edge2 = edges
  5613                                           
  5614         4        116.0     29.0      0.2              if edge in edges:
  5615                                                           pass
  5616                                           
  5617         8      10305.0   1288.1     15.1              chamfer_builder.AddChamfer(
  5618         4        142.0     35.5      0.2                  TopoDS.Edge_s(edge1.wrapped),
  5619         4         70.0     17.5      0.1                  TopoDS.Edge_s(edge2.wrapped),
  5620         4          9.0      2.2      0.0                  distance,
  5621         4          7.0      1.8      0.0                  distance2,
  5622                                                       )
  5623                                           
  5624         1         68.0     68.0      0.1          chamfer_builder.Build()
  5625                                           
  5626         1       5983.0   5983.0      8.8          return self.__class__(chamfer_builder.Shape()).fix()
  ```

anb, 2024-01-11:
Here you go.

I really think it's the FindFromIndex that is slow.

```
Timer unit: 1e-09 s

Total time: 1.32924 s
File: /Users/brett/dev/env-build123d/lib/python3.10/site-packages/build123d/topology.py
Function: _entities_from at line 2105

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
  2105                                               def _entities_from(
  2106                                                   self, child_type: Shapes, parent_type: Shapes
  2107                                               ) -> Dict[Shape, list[Shape]]:
  2108         1       3000.0   3000.0      0.0          res = TopTools_IndexedDataMapOfShapeListOfShape()
  2109                                           
  2110         2       5000.0   2500.0      0.0          TopExp.MapShapesAndAncestors_s(
  2111         1          0.0      0.0      0.0              self.wrapped,
  2112         1          0.0      0.0      0.0              inverse_shape_LUT[child_type],
  2113         1       1000.0   1000.0      0.0              inverse_shape_LUT[parent_type],
  2114         1          0.0      0.0      0.0              res,
  2115                                                   )
  2116                                           
  2117         1          0.0      0.0      0.0          out: Dict[Shape, list[Shape]] = {}
  2118         5       4000.0    800.0      0.0          for i in range(1, res.Extent() + 1):  
  2119         8 1329209000.0    2e+08    100.0              out[Shape.cast(res.FindKey(i))] = [
  2120         4      19000.0   4750.0      0.0                  Shape.cast(el) for el in res.FindFromIndex(i)
  2121                                                       ] 
  2122                                                       
  2123         1          0.0      0.0      0.0          return out

Total time: 1.32996 s
File: /Users/brett/dev/env-build123d/lib/python3.10/site-packages/build123d/topology.py
Function: chamfer_2d at line 5572

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
  5572                                               def chamfer_2d(
...
  5597         1      14000.0  14000.0      0.0          chamfer_builder = BRepFilletAPI_MakeFillet2d(self.wrapped)
  5598         1 1329258000.0    1e+09     99.9          edge_map = self._entities_from(Vertex.__name__, Edge.__name__)
  5599                                           
  5600         5          0.0      0.0      0.0          for vertex in vertices:
  5601         4       8000.0   2000.0      0.0              edges = edge_map[vertex]
  5602         4       2000.0    500.0      0.0              if len(edges) < 2:
  5603                                                           raise ValueError("Cannot chamfer at this location")
  5604                                           
  5605         4          0.0      0.0      0.0              if edge:
  5606                                                           if edge not in edges:
  5607                                                               raise ValueError("One or more vertices are not part of edge")
  5608                                           
  5609                                                           edge1 = edge
  5610                                                           edge2 = [x for x in edges if x != edge][0]
  5611                                           
  5612                                                       else:
  5613         4          0.0      0.0      0.0                  edge1, edge2 = edges
  5614                                           
  5615         4       4000.0   1000.0      0.0              if edge in edges:
  5616                                                           pass
  5617                                           
  5618         8     398000.0  49750.0      0.0              chamfer_builder.AddChamfer(
  5619         4       3000.0    750.0      0.0                  TopoDS.Edge_s(edge1.wrapped),
  5620         4       5000.0   1250.0      0.0                  TopoDS.Edge_s(edge2.wrapped),
  5621         4          0.0      0.0      0.0                  distance,
  5622         4          0.0      0.0      0.0                  distance2,
  5623                                                       )
  5624                                           
  5625         1       1000.0   1000.0      0.0          chamfer_builder.Build()
  5626                                           
  5627         1     263000.0 263000.0      0.0          return self.__class__(chamfer_builder.Shape()).fix()
```

jdegenstein, 2024-01-11:
OK, thank you for testing. My read on the situation after digging pretty deep into this issue is that there is some "overhead" when transferring objects between C++ (OCP kernel) and Python. This makes very simple operations for certain data structures take a lot longer than they should (even on x86). My untested suspicion is that apple silicon machines have even worse optimizations for this scenario for some reason. This particularly seems to be a problem for OCP data map structures which make use of pybind11 `stl.h` and I believe OCP uses that approach. https://github.com/CadQuery/pywrap/blob/master/bindgen/template_main.j2

Further reading here: https://pybind11.readthedocs.io/en/stable/advanced/cast/stl.html
and also this article here: https://zpz.github.io/blog/speeding-up-python-with-cpp-and-pybind11/

So, where does this leave us? It is unclear how easily this can be fixed by build123d. My best idea at the moment is to (try to) reduce the number of "data transfers" between OCP and build123d. To that end I created a new branch that needs testing by an apple silicon user.

Comparison to mainline: https://github.com/gumyr/build123d/compare/dev...jdegenstein:build123d:improve_chamfer2d

Installable via pip like: `pip install git+https://github.com/jdegenstein/build123d@improve_chamfer2d` the speed is about the same on x86 as the current version (maybe 10% slower). What I need is for someone to test the above benchmark on this new branch.

anb, 2024-01-12:
I tried your branch but unfortunately it doesn't make a huge difference : 

```
Total time: 1.69199 s
File: /Users/brett/dev/env-cadquery-test/lib/python3.10/site-packages/build123d/topology.py
Function: _entities_from at line 2105

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
  2105                                               def _entities_from(
  2106                                                   self, child_type: Shape, parent_type: Shape
  2107                                               ) -> Dict[Shape, list[Shape]]:
  2108         1       3000.0   3000.0      0.0          res = TopTools_IndexedDataMapOfShapeListOfShape()
  2109                                           
  2110         2       4000.0   2000.0      0.0          TopExp.MapShapesAndAncestors_s(
  2111         1          0.0      0.0      0.0              self.wrapped,
  2112         1          0.0      0.0      0.0              inverse_shape_LUT[child_type.__name__],
  2113         1       1000.0   1000.0      0.0              inverse_shape_LUT[parent_type.__name__],
  2114         1          0.0      0.0      0.0              res,
  2115                                                   )
  2116                                           
  2117         1     118000.0 118000.0      0.0          children = [e for e in self.shapes() if isinstance(e, child_type)]
  2118                                           
  2119         2 1691834000.0    8e+08    100.0          out: Dict[Shape, list[Shape]] = {
  2120                                                       children[idx]: [Shape.cast(el) for el in parents]
  2121         1      26000.0  26000.0      0.0              for idx, parents in enumerate(res)
  2122                                                       }
  2123                                           
  2124         1          0.0      0.0      0.0          return out
```

I stumbled upon this : https://github.com/tpaviot/pythonocc-core/blob/37ec91c0e7dfb6da4aa13240a86b9740ddaf07a6/src/Extend/TopologyUtils.py#L318C1-L318C1

It seems that OCC has an iterator usable on the result of TopTools_IndexedDataMapOfShapeListOfShape but I don't know if that makes it faster (avoids the conversions). Unfortunately, it's not in the Cadquery bindings but I might try to install the python-occ thing and run some tests.

jdegenstein, 2024-01-14:
@anb Ok, thank you for checking. In the meantime a much better solution was found to this issue and it was merged already. Can you test on the latest development version again? https://github.com/gumyr/build123d/pull/476

I am going to go ahead and close this issue since benchmarking was already performed.

By the way OCP 7.7.2 has many performance enhancements over OCP 7.7.1 which seems particularly important on aarch64 systems.

closed 2024-01-14 by jdegenstein

anb, 2024-01-15:
I tested branch dev and can confirm bottleneck completely gone 👍

Thanks also to @MatthiasJ1