gumyr/build123d #370: Get dimensions of face

Spectre5, 2023-11-08:
I'm selecting a face on a part that I know will be rectangular, but could be of unknown size.  I'd like to draw a circle on this face where the dimension of the circle is based on being "x" less than the minimum dimension of the rectangle.  What is the easiest way to achieve this?

```python
import build123d as cad
box = cad.Box(100, 80, 10)
top = box.faces().sort_by()[-1]
# create a circle with diameter 10 less than the minimum rectangle dimension (with or height)
# in this example, it would simply be a diameter of 70, but how would I do this if I didn't know the face dimensions?
```

gumyr, 2023-11-08:
There are dedicated `length` and `width` methods for this or alternatively one can use the Face's bounding box as shown here:
```py
import build123d as cad
from ocp_vscode import show_all

box = cad.Box(100, 80, 10)
top = box.faces().sort_by()[-1]
# create a circle with diameter 10 less than the minimum rectangle dimension (with or height)
# in this example, it would simply be a diameter of 70, but how would I do this if I didn't know the face dimensions?

print(top.geometry)  # RECTANGLE
circle_diameter = min(top.length, top.width) - 10
print(circle_diameter)  # 70
# or
circle_diameter = min(top.bounding_box().size.X, top.bounding_box().size.Y) - 10
print(circle_diameter)  # 70
hole = cad.Pos(top.center()) * cad.Circle(radius=circle_diameter / 2)

# Or in Builder mode
with cad.BuildPart() as bp:
    cad.Box(100, 80, 10)
    top = bp.faces().sort_by(cad.Axis.Z)[-1]
    circle_diameter = min(top.length, top.width) - 10
    with cad.BuildSketch(top):
        cad.Circle(radius=circle_diameter / 2)
    cad.extrude(amount=-1, mode=cad.Mode.SUBTRACT)

show_all()
```
[image]

Spectre5, 2023-11-09:
Awesome, thanks - I thought I looked everywhere in the docs but I obviously missed this.  I appreciate the response.

As a follow-up, is there an easy way to know how a plane derived from a face will be oriented?  Consider the code below, which prints `Plane(o=(0.00, -0.00, 35.00), x=(0.80, 0.60, -0.00), z=(-0.00, -0.00, -1.00))` for me.  I would have expected this plane to still have "x" oriented in (1,0,0), or maybe (-1,0,0), but it looks like x instead points towards one of the corners of the face.  Is that expected here?  If so, is there a rule to know how a face will be oriented ahead of time?

```python
import build123d as cad
from ocp_vscode import show_object
sketch = cad.Sketch() + [
    cad.Rectangle(100, 80),
    cad.Plane(origin=(0, 0, 40)) * cad.Rectangle(80, 60),
]
solid = cad.loft(sketch, ruled=True)
bottom = solid.faces().sort_by()[0]
shell = cad.offset(solid, amount=-5, openings=bottom)

face = shell.faces().sort_by()[-2]
outer = cad.make_face(cad.offset(face.wires(), amount=-2))
inner = cad.make_face(cad.offset(face.wires(), amount=-10))
shell += cad.extrude(outer - inner, -5)

face = shell.faces().group_by()[-2].sort_by_distance((0,0,0)).first
print(cad.Plane(face))

show_object(shell)
show_object(face)
```

[image]

gumyr, 2023-11-09:
The short answer is no.  A `Plane` defines a coordinate system where the origin and directions of all three axes are defined.  A planar `Face` doesn't have enough information to fully define a coordinate system, the x and y axes are free to rotate about the z axis (as long as they are 90 degrees apart and follow the right-hand-rule). However, the internal OpenCascade algorithm for this is usually quite good at picking sensible directions - but not always.

However, it is possible to create a custom `Plane` as follows: `Plane(origin=face.center(), x_dir=(1,0,0), z_dir=face.normal_at())` which allows the user to control the x axis.

closed 2023-12-03 by gumyr