gumyr/build123d #373: No matching distribution found for cadquery-ocp

MurKit, 2023-11-09:
Ubuntu 20
Python 3.11.5
pip 23.2.1 from /home/me/miniconda3/lib/python3.11/site-packages/pip (python 3.11)

```
python3 -m pip install git+https://github.com/gumyr/build123d
...
ERROR: Could not find a version that satisfies the requirement cadquery-ocp~=7.7.1 (from build123d) (from versions: none)
ERROR: No matching distribution found for cadquery-ocp~=7.7.1

```

_Other issues that did not help me:_

`pip install numpy-stl` didn't help
https://github.com/gumyr/build123d/issues/289

looks like the update doesn't work for me here
https://github.com/gumyr/build123d/pull/333

maybe related to apple silicon somehow
https://github.com/gumyr/build123d/pull/330

gumyr, 2023-11-09:
If you using a Mac with Apple silicon you need to follow the instructions listed here: https://build123d.readthedocs.io/en/latest/installation.html#special-notes-on-apple-silicon-installs

gumyr, 2023-11-09:
If you're running Ubuntu 20 there may be issues - IIRC there was a need for some custom settings.  Are you able to use a more up-to-date version of Ubuntu?  I use `Ubuntu 23.04`.

MurKit, 2023-11-09:
Sorry didn't mention it - I use an AMD and not apple.

Yeah I guess updating the system looks like an easier option.
I was just trying to run with what I have now.

jdegenstein, 2023-11-09:
Jumping in here late, you should be able to install cadquery-ocp (aka ocp) from conda-forge on Ubuntu 20.04 -- for some reason the manylinux version was increased on the pypi packages which broke compatibility with Ubuntu 20.04 (as you noticed with our readthedocs builds).

MurKit, 2023-11-09:
conda is so broken, couldn't install ocp with it, oh well

gumyr, 2023-11-10:
Should there be a note in the installation instructions that Ubuntu 20 isn't supported or at least not advised?

jdegenstein, 2023-11-10:
No, because I am pretty sure that a conda install of OCP is still possible on Ubuntu 20.04 -- might call for custom commands section in the docs? Maybe we should petition CQ for adjusting the pypi manylinux support back down to 2.31 (from 2.35)? 
https://pypi.org/project/cadquery-ocp/7.7.1/#files

and
https://pypi.org/project/cadquery-ocp/7.7.0/#files

gumyr, 2023-11-11:
@jdegenstein you could try to ask Adam to go back but I would guess there's a reason for requiring the upgrade. Ubuntu 20.04 LTS is being supported until 2025 so it would be unfortunate if we can't support it.

gumyr, 2025-05-15:
@jdegenstein this one is quite old - can we close it?

closed 2025-05-15 by jdegenstein