gumyr/build123d #376: Raise error on unknown kwargs

Spectre5, 2023-11-10:
I would recommend the build123d raise an error on unknown kwargs for the overloaded functions, or at least for the Pos class.  Multiple times in my short time using build123d I've spent longer than I'd like to admit trying to figure out why my position was not applying only to eventually realize that I need to use `Pos(Z=1)` and not `Pos(z=1)` (lower case).  If you don't want to raise an error, then perhaps we can at least allow upper or lower case letters for `Pos`?

Spectre5, 2023-11-10:
This is also inconsistent as `Rot` takes lower case letters x, y, and z.

gumyr, 2023-11-10:
Thanks this needs to be fixed.
```py
Pos(X, Y, Z)
Rot(x, y, z)
Vector(x, y, z)
Rotation(about_x, about_y, about_z)
Vertex(X, Y, Z)
make_wedge(delta_x, delta_y, delta_z, min_x, min_z, max_x, max_z)
Axis.X|Y|Z
Plane.XY|XZ,...
```
To be consistent all the x, y, z's should be replaced with X, Y, Z but does `about_X` make sense? Maybe these should be X, Y, Z as well.

Raising an error could be done with:
```py
    oops = ", ".join(set(kwargs.keys()).difference(["X", "Y", "Z"]))
    if oops:
        raise ValueError(f"Unexpected argument(s) {oops}")
```

gumyr, 2023-11-12:
Done.  All of these interfaces accept kwargs and raise an exception for unknown values.

closed 2023-11-12 by gumyr

Spectre5, 2023-11-12:
Great, thanks!