gumyr/build123d #384: Performance of Builder API relative to Algebra API is impacted by `fuse()` calls

fischman, 2023-11-15:
`Builder._add_to_context` [calls](https://github.com/gumyr/build123d/blob/ebc4f9b7af3e41e9f2a9aafbabfcf13b177e1155/src/build123d/build_common.py#L370-L379) `fuse` for every object that is created inside the builder's context, which can be quite expensive as shown in https://github.com/gumyr/build123d/commit/52a00fb6a77045aa85c95d34b41cffbe4aa51dad (where `s/PRIVATE/ADD/` makes the test fail because the Builder API takes ~20x longer than the Algebra API to do the same thing). Probably noteworthy that `Builder._add_to_context` is basically doing what https://build123d.readthedocs.io/en/latest/algebra_performance.html recommends not to do (prematurely calling `fuse`, and doing so repeatedly).

I don't know the codebase well enough yet to know if there's an easy solution.

This came up while investigating #383.

gumyr, 2023-11-15:
I'll look into the potential TechnicalDrawing problem but in general I don't see this - if you have an example it would be very helpful.  The code in `BuildCommon` that does the fuse is this:
```py
                        self._obj = self._obj.fuse(*typed[self._shape])
```
so it's a single fuse operation for all of the objects of the same type.

When I try this test:
```py
from build123d import *
from ocp_vscode import *
import timeit

start_time = timeit.default_timer()
with BuildSketch() as test1:
    with GridLocations(10, 10, 40, 40):
        RegularPolygon(6, 8)
print(f"Builder Time: {timeit.default_timer() - start_time:0.3f}s")

start_time = timeit.default_timer()
test2 = Sketch() + GridLocations(10, 10, 40, 40) * RegularPolygon(6, 8)
print(f"Algebra Time: {timeit.default_timer() - start_time:0.3f}s")

show(test1, test2)
```
these are the times:
```
Builder Time: 7.356s
Algebra Time: 6.135s
```
where the Builder is doing more work (like keeping track of new edges, etc) so I expect it would take a little longer.

[image]

fischman, 2023-11-15:
> if you have an example it would be very helpful

My example was linked in my original report (https://github.com/gumyr/build123d/commit/52a00fb6a77045aa85c95d34b41cffbe4aa51dad) but to inline it (and rephrase to match your example):
```py
from build123d import *
from ocp_vscode import *
import timeit

start_time = timeit.default_timer()
with BuildPart() as test1:
    [Box(1, 1, 1) for _ in range(20)]
print(f"Builder Time: {timeit.default_timer() - start_time:0.3f}s")

start_time = timeit.default_timer()
test2 = [Box(1, 1, 1) for _ in range(20)]
print(f"Algebra Time: {timeit.default_timer() - start_time:0.3f}s")

show(test1, test2)
```

This emits:
```
Builder Time: 0.149s
Algebra Time: 0.007s
```

Something I didn't realize until just now putting the above together is that the difference goes away if the `test2` line above is replaced with `test2 = Part() + [Box(1, 1, 1) for _ in range(20)]` which is more comparable to what the Builder API emits, though arguably this is not always/frequently needed when using the Algebra API.

fischman, 2023-11-15:
Not sure if silly or serious, but: is it reasonable (compatible w/ Builder's API) to postpone the fuse until `.part` is called?

MatthiasJ1, 2023-11-15:
In your example @fischman you are comparing creating 20 boxes vs creating + taking the union of 20 boxes.

Here is a different example:

```py
import timeit
from build123d import *
from bd_warehouse.thread import IsoThread

from ocp_vscode import show_object

major_diameter = 6
length = 8

starttime = timeit.default_timer()

with BuildPart() as nut:
    with BuildSketch():
        RegularPolygon(major_diameter * 0.75, 6)
        Circle(major_diameter / 2, mode=Mode.SUBTRACT)
    extrude(amount=length)
nut = nut.part.fuse(IsoThread(major_diameter, 1, length, False, "right", ("chamfer", "fade")))

elapsed_time = timeit.default_timer() - starttime
print(f"Outside Builder: {elapsed_time:.3f}s")

starttime = timeit.default_timer()

with BuildPart() as nut:
    with BuildSketch():
        RegularPolygon(major_diameter * 0.75, 6)
        Circle(major_diameter / 2, mode=Mode.SUBTRACT)
    extrude(amount=length)
    IsoThread(major_diameter, 1, length, False, "right", ("chamfer", "fade"))

elapsed_time = timeit.default_timer() - starttime
print(f"Inside Builder: {elapsed_time:.3f}s")
```

```
Outside Builder: 1.812s
Inside Builder: 8.408s
```