gumyr/build123d #385: Face added to part gives different results when added to BuildPart vs BuildSketch

MatthiasJ1, 2023-11-15:
```py
from build123d import *

slot1 = SlotOverall(45, 38)
slot2 = SlotOverall(60, 4)

with BuildPart() as p1:
    with BuildSketch(): add(slot1)
    with BuildSketch(Plane.XY.offset(100)): add(slot2)
    loft()

with BuildPart() as p2:
    add(slot1)
    with BuildSketch(Plane.XY.offset(100)): add(slot2)
    loft()

```
[image]

gumyr, 2023-11-15:
`BuildSketch` automatically orients all `Faces` so their normal's are "up" - in `p2` `add(slot1)` doesn't do this so the normal is "down". The `loft` algorithm is sensitive to `Face` normals so the results are different because `loft` is working with different `Faces`.  This is one of the weird things that the builders try to hide from the user but you've worked around that.

MatthiasJ1, 2023-11-15:
I checked and all of the normals are up. Either way, to be sure, I tried all combinations of original and flipped face normals and the outcome is unchanged.