gumyr/build123d #388: Integrating build123

gsohler, 2023-11-17:
Hi, i am guenther and i am the author of the openscad fork understanding python natively.
check out http://pythonscad.org/
I like very much the idea of build123 to compile 2d shapes from 1d and to compile 3d shapes from 2d shapes.
as my openscad understands python, it might work. 
just curious about the interfaces. how does buidl123 output its result?

Thank you and best regards

gumyr, 2023-11-17:
Hi guenther, I'm Roger Maitland (gumyr) and build123d is my project. I first saw your project yesterday - being able to use OpenSCAD in Python instead of writing complex stuff with recursion (which I had to do with many of my OpenSCAD projects) is a big improvement.

build123d is a Python API to the OpenCascade BREP CAD kernel much like your Python API is to the CGAL mesh CAD kernel (directly or indirectly).  There are many ways to output results in build123d, there are exporters to popular CAD file formats (described here: https://build123d.readthedocs.io/en/latest/import_export.html) including mesh formats like STL and 3MF.  The viewers (like https://github.com/bernhard-42/vscode-ocp-cad-viewer) first "tessellate" the build123d objects into a mesh format for display.

Are you thinking about creating similar objects to the ones described here:https://build123d.readthedocs.io/en/latest/objects.html but with OpenSCAD implementations?

I'm happy to answer any questions you have, just let me know how I can help.

Cheers,
Roger

gsohler, 2023-11-17:
Hi Roger,

Thank you for your valuable reply
In the meantime I have found  the export_stl function, (not understanding
the .part. thing yet and I am able
to display the stl in openscad

Right now I am performing the Selector tutorial  and the command sequence
somehow reminds me on RPN ( reverse polish notation)
But the topic is exciting, It shows alternative ways to compose objects.

I think I would like to put these codes directly into the openscad editor
without the additional save_stl and load-openscad step.
There is a function polyhedron in openscad which can accept an object
defined by points and triangles.

The vscode is not yet understood. does it mean "Visual Studio"? I am not a
windows user primarily ...

Yes, I would like to display all the build123 stuff in openscad ( all
solids it can potentially generate)
openscad can display 3D and 2D stuff ... but I would even extend openscad
to display 1D stuff when reasonable ...

Can you tell me, which are the exact additional dependencies when I use
build123 ?

need to drive home now...
have a nice evening

jdegenstein, 2023-11-17:
@gsohler check out this page from the docs on "editors and viewers" https://build123d.readthedocs.io/en/latest/external.html

build123d is technically just the python library and does not come with an editor or a viewer. There are two editor options; (1) VSCode which is cross platform (incl Linux) and available for free, (2) my fork of CQ-editor which has enhanced build123d support. (1) is the recommended option, but (2) has static executables with all needed dependencies bundled so it is typically very easy to just download and run.

gumyr, 2023-11-17:
As @jdegenstein mentioned, vscode refers to Microsoft's Visual Studio Code (https://code.visualstudio.com/) which is a very popular editor and IDE which is available on Windows, Mac and Linux (I develop in Linux). The ocp_vscode viewer runs within the VSC IDE providing the visual feedback while the Python code is just standard Python so all of the capabilities of the IDE apply, even Debug mode where the user can visually single step through their code - here is an example: [image].  

To get a better idea of the environment, here is a video of @jdegenstein "speed racing" CAD: https://www.youtube.com/watch?v=fH8aW27jEiw - see https://build123d.readthedocs.io/en/latest/tttt.html for more info on speed racing.

build123d can be used two ways:
1) Direct manipulation of the CAD objects with little structure called Algebra mode which looks like this:
```py
ex2 = Box(length, width, thickness)
ex2 -= Cylinder(center_hole_dia / 2, height=thickness)
```
2) Builder mode which puts more structure into the code and provides some handy features not available in Algebra mode which looks like this (resulting in the same object):
```py
with BuildPart() as ex2:
    Box(length, width, thickness)
    Cylinder(radius=center_hole_dia / 2, height=thickness, mode=Mode.SUBTRACT)
```
In this case `ex2` refers to an instance of the `BuildPart` builder; to get the part back from this builder one refers to it with the `part` attribute - `ex2.part` in this case.  `BuildSketch` and `BuildLIne` have `sketch` and `line` attributes to get at their objects.

To directly transfer build123d objects to OpenSCAD a polyhedron could be created (for solid objects) which is how I created this "rope" model in OpenSCAD (https://www.thingiverse.com/thing:4242723) 
[image]

The build123d `tessellate` method produces `tuple[list[Vector], list[tuple[int, int, int]]]` which I believe is exactly the format of points and triangles required by the OpenSCAD `polyhedron` so it's probably quite easy to create a single function/method that would translate solid objects to OpenSCAD.

The 1D and 2D objects could be converted to points or point arrays for OpenSCAD if that is of interest.

Hopefully this is helpful.

gumyr, 2023-11-17:
Here is a link to the docs for `tessellate`: https://build123d.readthedocs.io/en/latest/direct_api_reference.html#topology.Shape.tessellate

gumyr, 2023-11-17:
The build123d pyproject.toml file has all of the dependencies: https://github.com/gumyr/build123d/blob/dev/pyproject.toml The big one is  "cadquery-ocp ~= 7.7.1" which is a Python wrapped version of the C++ OpenCascade CAD kernel.

gsohler, 2023-11-17:
Thank you for all this information!
I cannot yet use it, will probably need later.
I was trying to boldly use it in my openscad and of course it failed:

https://imgur.com/a/3MGHHGc

it appears that the build123d module installs into ~/.local/lib/python/site-packages
whereas my python mainly runs ins /usr/lib/python3.13/site-packages/

how can i change installation procedure to install build123d globally ?
Also there is a mess in my place between python3.11 and python3.13
any clues on the error shown ?

jdegenstein, 2023-11-17:
I highly recommend you use either a venv or a conda environment. I also do not know if build123d will work above python3.11 at the moment. Regarding your screenshot, I believe this is a known issue with typing_extensions==4.4.0 see issue here https://github.com/python/typing_extensions/issues/243

@gumyr I believe we need to change the `pyproject.toml` to `"typing_extensions >= 4.6.0, <5",` I will create a PR for it now.

jdegenstein, 2023-11-18:
@gsohler it might be worth trying again, as the issue you were seeing should be fixed now (https://github.com/gumyr/build123d/pull/389). I still recommend using py3.11 but it might work now on py3.12.

gsohler, 2023-11-18:
Hi Jdegenstein,

- I was  already able to progress by manually applying the change which you suggested to do a PR  yesterday, but I am stuck somewhere else.

Is it possible for some direct chat without the build123d issue log ?

jdegenstein, 2023-11-18:
@gsohler Best way would be to join our discord (https://discord.com/invite/Bj9AQPsCfx) which is also linked from the README.md

gumyr, 2023-11-30:
@gsohler, given your progress and the discussions on Discord, can this Issue be closed?

closed 2023-11-30 by gumyr