gumyr/build123d #391: GridLocations give different results when used inside of context vs for loop

MatthiasJ1, 2023-11-21:
According to the ReadTheDocs, both of these are supported uses of Locations yet they give different results.

```py
with BuildPart() as p:
    Box(50, 10, 1)
    with Locations((-5, 0, 0)):
        with GridLocations(10, 0, 5, 1):
            Box(1, 10, 4)
```
[image]

```py
with BuildPart() as p:
    Box(50, 10, 1)
    with Locations((-5, 0, 0)):
        for l in GridLocations(10, 0, 5, 1):
            Box(1, 10, 4, l)

####### or #######

with BuildPart() as p:
    Box(50, 10, 1)
    with Locations((-5, 0, 0)):
        for l in GridLocations(10, 0, 5, 1):
            with Locations(l):
                Box(1, 10, 4)
```
[image]

gumyr, 2023-11-30:
Nested Locations are impacted by other Locations within their scope so in your examples (non Sketch - I haven't looked into that yet) you're nesting 3 Locations so the first one is being applied twice.  If you move the GridLocations outside of the scope of the 1st Locations (e.g. outside of any of the `with` blocks) I believe you'll get the results you're looking for.

gumyr, 2023-12-03:
I don't understand what you're proposing here - maybe some code would help.  The current behaviour is as follows:
```py
with BuildPart() as plate:
    Box(120, 50, 10)
    with Locations(plate.faces() >> Axis.Z):
        with GridLocations(70, 0, 2, 1):
            with PolarLocations(10, 4):
                CounterBoreHole(2, 4, 2)
```
[image]

How would your proposal be different in this case?

gumyr, 2023-12-11:
With respect to the following:
```py
with BuildSketch() as A:
    with Locations((0,0), *GridLocations(5,5,2,2)):
        Circle(1)

with BuildSketch(Pos(10, 0)) as B:
    with Locations((0,0), *GridLocations(5,5,2,2)):
        Circle(1)
```
I don't know of any way for the S/W to know what the use intent is so it's going to have to stay the same.  Here how this should be done:
```py
with BuildSketch() as A:
    with Locations((0,0), *GridLocations(5,5,2,2)):
        Circle(1)

with BuildSketch(Pos(10, 0)) as B:
    with Locations((0,0), *GridLocations(5,5,2,2).local_locations):
        Circle(1)
```
[image]

In this case `local_locations` should be used as they are relative to Workplane and will be adjusted within the Locations call.  `for` loops are definitely 2nd class citizens here as the optimized case is for Locations in `with` blocks.  I've updated the docs to show the `local_locations` attribute.

I've created an Issue (https://github.com/gumyr/build123d/issues/431) to explore if's correct and possible to change what the Location classes return based on if they are in `with` blocks.

closed 2023-12-11 by gumyr