gumyr/build123d #398: BuildPart's workplanes do not affect nested BuildSketch

MatthiasJ1, 2023-11-27:
```py
with BuildPart(Plane.XZ) as p:
    Cylinder(1, 5)
    with BuildSketch() as s:
        Circle(1)
    extrude(amount=5)
```

[image]

gumyr, 2023-12-02:
This is by design. There is no attempt to inherent workplanes from nested objects as there could be many levels of nesting with sub-classing and use within functions which could be very confusing for the user.

MatthiasJ1, 2023-12-02:
This would contradict that:

```py
with BuildPart() as p:
    with BuildSketch(Plane.XY.shift_origin((3,))) as s:
        with BuildLine():
            CenterArc((0,0), 1, 0, 360)
        make_face()
    extrude(amount=5)
```
[image]

closed 2023-12-02 by gumyr