gumyr/build123d #419: duration of fillet is long due to is_valid

Ruudjhuu, 2023-12-04:
When performing a fillet on a complex object in the later stages, it takes a very long time due to the `is_valid` check.

In the first image you can see that the program spends 51% of the time in `is_valid`. (1.12 of a 2.21 execution). The second image shows that `is_valid` is executed in the fillet function. I do not know if it is possible to optimize this (different check?, toggable?). maybe it is an issue in the OCP python bindings. I had a quick look at the FreeCad code, they do exactly the same (but in C++). If I export my model and do the fillet in FreeCad, it is way quicker (under 10 sec instead of minutes). 

I am aware this issue lacks a lot of information, it is just an observation. If you think it is worth looking into, I can provide more info but I will also accept a "won't fix" label. 

[image]

[image]

profiler tool: 
https://www.speedscope.app/
profiler file:
[profile.speedscope.tar.gz](https://github.com/gumyr/build123d/files/13551079/profile.speedscope.tar.gz)

gumyr, 2023-12-05:
`is_valid` is essentially an OCP method so there isn't much that I can do about that. Are you using an Apple Silicon Mac? There have been some other slow-downs on Macs that I don't understand.

Ruudjhuu, 2023-12-06:
No, I am not using an apple silicone.

I'll close this issue as it is not related to this project.

closed 2023-12-06 by Ruudjhuu