gumyr/build123d #432: Bug with the 3MF exporter

KarimJedda, 2023-12-11:
Hello! Thanks for the amazing library, it's a lot of fun :) 

Here's an issue I'm struggling with. The following, doesn't work: 

```python
from build123d import *

with BuildPart() as funky_box:
    Box(3, 3, 3)
    with BuildSketch(*funky_box.faces()):
        Rectangle(1, 2, rotation=45)
    extrude(amount=10)

outset = funky_box.part
outset.color = Color("blue")  

exporter = Mesher(unit=Unit.CM)
exporter.add_shape(outset)
exporter.write("single.3mf")
```
with the following error message: 
```
Traceback (most recent call last):
  File "/home/potato/tests/build123d/third.py", line 14, in <module>
    exporter.add_shape(outset)
  File "/home/potato/tests/build123d/venv/lib/python3.10/site-packages/build123d/mesher.py", line 422, in add_shape
    raise RuntimeError("3mf mesh is invalid")
RuntimeError: 3mf mesh is invalid
```

whereas the following, does work:

```python
from build123d import *

with BuildPart() as funky_box:
    Box(3, 3, 3)
    with BuildSketch(*funky_box.faces()):
        Rectangle(2, 2, rotation=45) # changed here
    extrude(amount=10)

outset = funky_box.part
outset.color = Color("blue")  

exporter = Mesher(unit=Unit.CM)
exporter.add_shape(outset)
exporter.write("single.3mf")
```

An `STL` export works in both cases without errors, so I suspect this to be an issue with `py_lib3mf`. Any help would be appreciated, meanwhile I'll stick to `STL`.

jdegenstein, 2023-12-11:
This may be related to https://github.com/gumyr/build123d/issues/350 and https://github.com/gumyr/build123d/issues/311

gumyr, 2023-12-15:
Thanks for raising the issue.  I was able to isolate the problem further based on your test:
```py
with BuildPart() as funky_box:
    Box(3, 3, 3)
    with BuildSketch(funky_box.faces().sort_by(Axis.Z)[-1]):
        Rectangle(1, 2)
    extrude(amount=1)
```
[image]

works, but
```py
with BuildPart() as funky_box:
    Box(3, 3, 3)
    with BuildSketch(funky_box.faces().sort_by(Axis.Z)[-1]):
        Rectangle(1, 2, rotation=45)
    extrude(amount=1)
```
[image]
fails. I have no idea why rotating the box on the face would make a difference but clearly it does. Rotation even works for angles <= 0.90 degrees.

gumyr, 2023-12-15:
Another clue to what is going on.  When the angle is set to 0.90 degrees (which works), the ocp and 3mf mesher produce the following sets lengths of vertices and triangles:
```
len(ocp_mesh_vertices_set)=16, len(vertices_3mf_set)=16
len(triangles_set)=28, len(triangles_3mf_set)=28 
```
while if the angle is changed to 0.91 degrees (the invalid case), one gets:
```
len(ocp_mesh_vertices_set)=20, len(vertices_3mf_set)=16
len(triangles_set)=28, len(triangles_3mf_set)=28
```
so 4 more unique vertices were created by the ocp mesher.  Maybe there is a way to find and remove these extra vertices?

gumyr, 2023-12-16:
A bit more information.  The difference between the rotation 0.90 and 0.91 cases is that OCP generates four duplicate vertices in the failing case.  The Mesher algorithm detects this situation, removes the duplicate vertices and modifies the triangle list appropriately.  The mesh for the 0.91 case is shown here:
[image]
which looks correct but neither passes the lib3mf `IsValid` nor `IsManifoldAndOriented` tests. However, when a b3d `Shell` is created from these `Faces`, both the `is_valid()` and `is_manifold` method/properties returns True. The `Solid` created from this `Shell` is also valid.

gumyr, 2024-01-07:
Thanks @KarimJedda, duplicate vertices due to precision was the problem.  Fixed.

closed 2024-01-07 by gumyr

ninja-, 2024-01-11:
I've just hit the same bug, but with an entirely different geometry than above examples.
But the fix still worked like magic. Thanks!