gumyr/build123d #436: Sphere edge tangent incorrect

MatthiasJ1, 2023-12-18:
```py
s = Sphere(10)
e = s.edge()
l = Location(e@0.7, e%0.7)
```

[image]

gumyr, 2023-12-18:
It looks correct to me:
```py
from build123d import *
from ocp_vscode import *

sphere_edge = Sphere(1.5).edge()
tangent = Line(sphere_edge @ 0.7, (sphere_edge @ 0.7) + (sphere_edge % 0.7) * 1.5)
loc = sphere_edge.location_at(0.7)

show(sphere_edge, tangent, loc)
```
[image]

MatthiasJ1, 2023-12-18:
Ah I assumed `Location(A:vector, B:vector)` would construct a Location with `position=A` and `normal=B` but instead it does `position=A` and `normal=euler_angles(B)`. I don't see a constructor for creating a Location with position and normal, what is the recommended method? `Mixin1D.location_at` uses direct OCCT functions. `Location(Plane(sphere_edge@0.7, zdir=sphere_edge.tangent_at(0.7)))` does not work either, but this may be a bug unless I am misinterpreting something again:

```py
p = Plane((0,0,0), xdir=(0,0,1), zdir=(1,0,0))
```

[image]

gumyr, 2023-12-20:
The problem with the Location from Plane is that Plane silently discards invalid kwargs (it's `x_dir` not `xdir`) and should be fixed:
```py
show(Location(Plane((0, 0, 0), x_dir=(0, 0, 1), z_dir=(1, 0, 0))))
```
[image]

@jdegenstein and I have been discussing changing the `Location.__init__` signatures as we agree they don't seem to represent the common use cases.

gumyr, 2023-12-20:
I've raised two new issues to address the problems described here:

- https://github.com/gumyr/build123d/issues/440
- https://github.com/gumyr/build123d/issues/441

and will close this issue.  Thanks for your dedication to making build123d better.