gumyr/build123d #441: Review the Location constructor and optimize the valid overloads

gumyr, 2023-12-20:
Some of the Location constructor's overloads seem confusing and may not be optimal.  Specifically,
```py
    @overload
    def __init__(
        self, translation: VectorLike, direction: VectorLike, angle: float
    ):
```
seems misleading at best.  The term `translation` and `direction` should also be reconsidered as there are `position` and `orientation` properties.

jdegenstein, 2024-03-24:
kwargs are currently not working in `Location` either `Location((1,2,3),rotation=(10,20,30))` returns `TypeError: Location.__init__() got an unexpected keyword argument 'rotation'`

jdegenstein, 2025-05-15:
@gumyr I have been looking at this issue again and here are the current overloads for the `Location` class. I propose we eliminate the two overloads marked with "X" below and combine marked with "C" below.
```py
class Location:
    "@overloads" #omitting future @overload for brevity
    def __init__(self):
    def __init__(self, location: Location):
X   def __init__(self, translation: VectorLike, angle: float = 0):
C   def __init__(self, translation: VectorLike, rotation: RotationLike | None = None):
C   def __init__(self, translation: VectorLike, rotation: RotationLike, ordering: Extrinsic | Intrinsic):
    def __init__(self, plane: Plane):
    def __init__(self, plane: Plane, plane_offset: VectorLike):
    def __init__(self, top_loc: TopLoc_Location):
    def __init__(self, gp_trsf: gp_Trsf):
X   def __init__(self, translation: VectorLike, direction: VectorLike, angle: float):
┕ DELETE those marked with "X", COMBINE marked with "C"

    #actual constructor \/
    def __init__(self, *args): # consider adding **kwargs
        ...
```
Furthermore I agree that all uses of `translation` should be replaced with `position` instead. Additionally I think that `rotation` should be renamed to `orientation` to match with the class properties. I also propose we also reorganize the overloads, as the order they are listed affects the order they show up in the language server. Here are my proposed, reworked overloads / constructor signature.

```py
class Location:
    "@overloads" #omitting future @overload for brevity
    def __init__(self, position: VectorLike, orientation: RotationLike | None = None, ordering: Extrinsic | Intrinsic | None = None):
    def __init__(self, plane: Plane):
    def __init__(self, plane: Plane, plane_offset: VectorLike):
    def __init__(self, location: Location):
    def __init__(self):
    def __init__(self, top_loc: TopLoc_Location):
    def __init__(self, gp_trsf: gp_Trsf):

    #actual constructor \/
    def __init__(self, *args, **kwargs):
        ...
```
What do you think? If this is acceptable I can work on the actual constructor init.

gumyr, 2025-05-17:
@jdegenstein thanks for all the work you did to close this one.  Commit 67115111e2139540fff5560fceaf7ade9a1c3ac8 adds a little to your PR #991 to handle all of the overload cases except Plane/plane offset so it has minimal backwards compatibility issues.  All of your kwargs tests pass in addition to the existing Location tests (minus the one Plane/plane offset test that isn't needed anymore).

Hopefully this issue is good and closed now.

closed 2025-05-17 by gumyr