gumyr/build123d #442: joints do not play well with located and moved (and as such with algebra mode)

bernhard-42, 2023-12-20:
I ran into an issue with joints and `located` or `moved`
```pthon
b1 = Solid.make_box(1, 1, 1)

l = Location((0.5, 0.5, 1), (0, 0, 0))
j1 = RigidJoint("j1", b1, joint_location=l)

b1 = b1.located(Rotation(10, 20, 30))

show(b1, j1.parent, j1.symbol, b1.joints["j1"].symbol)
```

Since `located` creates a deep copy, `j1` will be decoupled from `b1.joints["j1"]`
[image]

When I use `locate` it works:
```python
b1 = Solid.make_box(1, 1, 1)

l = Location((0.5, 0.5, 1), (0, 0, 0))
j1 = RigidJoint("j1", b1, joint_location=l)

b1 = b1.locate(Rotation(10, 20, 30))

show(b1, j1.parent, j1.symbol, b1.joints["j1"].symbol)
```
[image]

Biggest impact: Joints don't play well with algebra mode (remember, `loc * shape` is implemented via `moved`)

gumyr, 2024-06-11:
@bernhard-42 I can't reproduce this problem.  Here is my example:
```py
from build123d import *
from ocp_vscode import show

before = Solid.make_box(1, 1, 1)
RigidJoint("j1", before, joint_location=Location((0.5, 0.5, 1), (0, 0, 0)))
before.color = Color("blue")

after = before.located(Rotation(0, 90, 0))
after.color = Color("red")

show(before, after, before.joints["j1"].symbol, after.joints["j1"].symbol)
```
[image]
The joint (which is relative to the part) moves with the part.  

In the original code `show(b1, j1.parent, j1.symbol, b1.joints["j1"].symbol)` is going to show the parent of `j1` which is not the joint attached to the `located` part anymore as it was copied too so that might have been confusing.

By the way, I was unable to get `render_joints=True` to work in my show command and displayed the symbols instead.

closed 2024-06-19 by gumyr

bernhard-42, 2024-06-20:
yes, you're right. After having worked on the joints improval I do understand joints much better now and using the variable `j1` in my example was the mistake. So all good here, thanks.

I am reworking the converter to ocp at the moment, so I can't easily test why `render_joints` is not working, but I'll ensure it'll work in the new version