gumyr/build123d #455: edges(Select.NEW) fails for chamfer

MatthiasJ1, 2024-01-04:
```py
with BuildPart() as p:
    Box(10,10,10)
    chamfer(edges(), 1)
    e = edges(Select.NEW)
    assert e
```
```
AssertionError
```

`topology.new_edges` "sequence of shapes" needs to account for `Mode`.

gumyr, 2024-01-05:
Maybe this is more of a documentation/expectations problem.  `Select.NEW` is designed to find new edges that were not in either of the Shapes being combined as follows:
```py
with BuildPart() as new_ex:
    Box(10, 10, 10)
    Cylinder(2, 20)
    n = edges(Select.NEW)
```
[image]
so that these new edges could be easily worked with (for example, filleted).  `Select.LAST` will select all of the edges changed by the last operation, as shown here for the above example:
[image]

Does this address the issue?

MatthiasJ1, 2024-01-05:
I might have missed it but I could't find documentation for `Select.NEW`. For the example below:

```py
with BuildPart() as part:
    Box(10, 10, 10)
    chamfer(edges()>>Axis.Z<<Axis.Y, 5)
    new = edges(select=Select.NEW)
    last = edges(select=Select.LAST)
```
I get the following. The edges under "new expected" did not exist in the object prior to the chamfer operation (although you could argue one of the |X edges was the original edge moved).
[image]

jwagenet, 2025-04-19:
This can be closed with the implementation of #959 as a documentation solution.

However, I agree the `Select.NEW` is confusing with fillet and chamfer (possibly other operations) and an exception for those ops to return "new expected" as in @MatthiasJ1's example.

Those edges can be found with `new_edges` (which is what `Select.NEW` uses under the hood):
````python
with BuildPart() as part:
    box = Box(10, 10, 10)
    chamfer(edges() >> Axis.Z << Axis.Y, 5)
    new = new_edges(box, combined=part)
````