gumyr/build123d #457: Add Align.None

MatthiasJ1, 2024-01-05:
When used, the created object would simply skip the step of aligning according to its bounding box. An example use would be when specifying coordinates with `Polygon`

gumyr, 2024-01-05:
This is already supported, albeit in a slightly different form: `Polygon((0, 0), (1, 0), (1, 1), align=None)` operates exactly as you describe and skips all the alignment code.  Unfortunately, this feature isn't well documented. The `Align` Enum could be expanded to include a `NONE` option as well - should there be redundant mechanisms for doing this?

MatthiasJ1, 2024-01-05:
Ah thanks, `align=None` fine, like you said it just needs to be documented. One way to do this could be:

```py
Align(Enum):
  CENTER=auto()
  ...
  NOALIGN=None
```

gumyr, 2024-10-13:
The type of `align` in the `BoundBox.to_align_offset` method is incorrect and should be fixed as part of this issue.