gumyr/build123d #459: Text does not factor in line height

MatthiasJ1, 2024-01-05:
The text alignment bounding box height should be determined by the font's line height rather than the rendered text's bounding box. The current implementation leads to errors like this:

```py
Text("A car.", 5)
```
[image]

```py
Text("A", 5) + Pos(6)*Text("car", 5) + Pos(10)*Text(".", 5)
```
[image]

gumyr, 2024-01-06:
Align applies to Text like all objects:
```py
t = (
    Text("A", 5, align=Align.MIN)
    + Pos(5) * Text("car", 5, align=Align.MIN)
    + Pos(12) * Text(".", 5, align=Align.MIN)
)
show(t)
```
[image]

The default is `Align.CENTER,Align.CENTER`.  Do you still think there is a problem here?

MatthiasJ1, 2024-01-06:
```py
t = (
    Text("ye", 5, align=Align.MIN)
    + Pos(5.5) * Text("s", 5, align=Align.MIN)
)
show(t)
```

[image]

gumyr, 2024-01-06:
If alignment worked based on the font's line height, vertically centering a single lower case letter would always be wrong which would be very frustrating.  build123d isn't a typesetting tool. Users are primarily going to be embossing very small amounts of text onto their parts so control over single characters is more important then sentences spanning multiple `Text` blocks. 

Won't fix.

MatthiasJ1, 2024-01-06:
@gumyr if you want to center text based on its bounding box rather than the text parameters you could easily realign the object using its bounding box (`add` could have an align parameter) or you could have an `alignmode=[Text, Geometry]` argument for `Text`.

Where this is an issue is say you have a ruler or panel with text underneath some feature. If the text includes different anchors ("mode" vs "43" vs "yaw") there is currently no way to align those without manually specifying offsets for each label.

gumyr, 2024-01-07:
Sorry for closing the issue prematurely.  I see now that there are some circumstances where the current implementation would make it difficult to align a series of Text elements as you describe.  I don't think just changing how align works is the correct solution though; some new functionality is required to introduce text specific alignment features.

CarpeNecopinum, 2024-09-21:
My workaround for this is the following function:

```py
def SafeText(text, size, *args, **kwargs):
    modded_text = f"█ {text} █"
    faces = Text(modded_text, size, *args, **kwargs)
    faces = faces.faces().sort_by(Axis.X)[1:-1]

    return faces
```

i.e. I add a U+2588 "Full Block" to either end of the String, and filter out the resulting two faces afterwards. That way, aligning (especially vertically) is a lot more consistent.

gumyr, 2024-09-22:
Unfortunately, the size of the "Full Block" isn't a very accurate indicator of where the descender bottoms out:
```py
text_abg = Text("█abg", font_size=50, align=Align.MIN)
```
[image]

What might be more deterministic is finding the "baseline" (i.e. the bottom of the "a") and having an option to `align` vertically based on this baseline as described above with a new parameter `alignmode=AlignMode.TEXT` or `alignmode=AlignMode.GEOMETRY`.  Would that solve this issue?

jwagenet, 2025-03-19:
In the absence of the actual font info since OCC handles the font, I'd probably suggest using a character like 'z' or 'm' which typically has a flat bottom on the baseline in *most* fonts. While characters like 'a' and 'o' sit on the baseline, rounded characters often extend over for optical reasons. Fonts are weird and vary quite a bit, so any consistent solution is fine.

I think using `AlignMode` is a great way to specify which behavior the user wants.

MatthiasJ1, 2025-03-19:
[image]
[image]

jwagenet, 2025-03-19:
I too could find numerous examples which don't work for probably any letter, hence "Fonts are weird and vary quite a bit, so any consistent solution is fine" absent actually opening the font file and finding the baseline in the emspace.

'o' might be best for a curved letter because it should rarely have strokes that descend, unlike 'a'. '.' (period) could be reliable, but may not exist in some decorative fonts. Some fonts may not have a full upper or lowercase set either, but a median of letters which normally land on the baseline might produce something. This doesn't even get into non-latin fonts (which could fall back on geometry alignment).

jwagenet, 2025-04-11:
Implementing vertical text alignment with the horizontal text alignment from #459 helps add more text control.

Funnily enough this issue was likely already resolved by using `align=(*, None)`, as the default text alignment from OCCT is left, bottom.