gumyr/build123d #460: pickling broken for STLs imported through Mesher

MatthiasJ1, 2024-01-05:
```py
from build123d import *
import pickle

part = Box(1,1,1) - Cylinder(0.3, 2)
part.export_stl("test.stl")
stl = Mesher().read("test.stl")

dump = pickle.dumps(part)
part = pickle.loads(dump)

dump = pickle.dumps(stl)
stl = pickle.loads(dump)
```

```
    stl = pickle.loads(dump)
          ^^^^^^^^^^^^^^^^^^
OCP.Standard.Standard_Failure: EXCEPTION in BinTools_ShapeSet::ReadGeometry(S,OS)
0x16bdbc190 : Standard_Failure: BinTools_SurfaceSet::ReadGeometry: UnExpected BRep_PointRepresentation = 131
```

jdegenstein, 2024-01-05:
Probably related to https://github.com/gumyr/build123d/issues/350

bernhard-42, 2024-01-06:
If it helps, this is not a OCCT problem:

```python
In [115]: from build123d import *
     ...: import pickle
     ...: 
     ...: part = Box(1,1,1) - Cylinder(0.3, 2)
     ...: part.export_stl("test.stl")
     ...: stl = Mesher().read("test.stl")

In [116]: from ocp_tessellate.ocp_utils import serialize, deserialize
In [117]: dump = serialize(stl[0].wrapped)
In [118]: stl2 = deserialize(dump)
In [119]: stl2
Out[119]: <OCP.TopoDS.TopoDS_Shape at 0x2a70fe770>
```

So the issue is hidden in the `pickle` code in persistence.py, maybe some encoding issues?

bernhard-42, 2024-01-06:
CC: @jdegenstein @Jojain

Jojain, 2024-01-06:
We might have found out this issue (on the work on measure tool of OCP CAD Viewer) after I proposed the PR to build123d.

As far as we know there is not really any other solution that doing what we did, i.e using the filesystem.

gumyr, 2026-10-01:
The pickle round trip works now.

closed 2026-10-01 by gumyr