gumyr/build123d #466: Face.geom_type() returns string rather than GeomType

MatthiasJ1, 2024-01-07:
`Face.geom_type()` might also make more sense as `@property`.

```py
print(f"{Rectangle(1,1).face().geom_type() = }")
print(f"{GeomType.PLANE = }")
print(f"{Rectangle(1,1).face().geom_type()==GeomType.PLANE = }")
```
```
Rectangle(1,1).face().geom_type() = 'PLANE'
GeomType.PLANE = <GeomType.PLANE>
Rectangle(1,1).face().geom_type()==GeomType.PLANE = False
```

gumyr, 2024-01-07:
Duplicate of Issue #321

jdegenstein, 2024-02-22:
Fixed with PR #559
```py
from build123d import *
print(f"{Rectangle(1,1).face().geom_type = }")
print(f"{GeomType.PLANE = }")
print(f"{Rectangle(1,1).face().geom_type==GeomType.PLANE = }")
```
now returns
```py
Rectangle(1,1).face().geom_type = <GeomType.PLANE>
GeomType.PLANE = <GeomType.PLANE>
Rectangle(1,1).face().geom_type==GeomType.PLANE = True
```

closed 2024-02-22 by jdegenstein