gumyr/build123d #468: Adding an unrelated solid will undo a split

MatthiasJ1, 2024-01-08:
The same issue occurs when using the Builder API.
```py
p = split(Box(1,1,1), keep=Keep.BOTH)
```
[image]

```py
p = split(Box(1,1,1), keep=Keep.BOTH)
p += Pos(0,2)*Sphere(0.5)
```
[image]

gumyr, 2024-01-08:
`BuildPart` is primarily intended to build a part so `Keep.BOTH` limited usefulness - although the split `Compound` object is returned so the user could save this result.  The `clean` operation is being applied when a new object is added to the part under construction which causes the previously `split` halves to merge together again.  If both halves are important, the `split` operation should be limited to the end of the `BuildPart` scope.

closed 2024-01-08 by gumyr

MatthiasJ1, 2024-01-09:
The same issue is observed even when all operations are called with `clean=False`