gumyr/build123d #469: pack leaves floating objects

MatthiasJ1, 2024-01-08:
The `pack`'s docstring states `Pack objects in a squarish area in Plane.XY.`
I believe the intended use for this function is to arrange objects to be sent for cutting or 3d printing. In both cases you would want the objects placed on the plane.

BTW, theres no mention of `pack` in the documentation.

```py
parts = [Box(1,1,1)] * 6
parts[0] = Pos(0,0,2) * parts[0]
parts = pack(parts, 0.5)
```
[image]

jdegenstein, 2024-01-08:
Pack is mentioned on this page https://build123d.readthedocs.io/en/latest/tips.html

MatthiasJ1, 2024-01-08:
Huh, the only thing that shows up when using the search bar is TTT Party Pack

gumyr, 2024-01-08:
Even though `pack.pack()` is in the docs in the above section the RTD search engine doesn't find "pack".  The documentation of this feature could certainly be improved.

erooke, 2024-11-17:
`align_z` was added in 9f189b0 which allows for this behavior. Should this be closed?

closed 2024-11-17 by gumyr