gumyr/build123d #480: offset with openings on a tapered cylinder (normal cup style shape) fails with BRep_API: command not done

philipsd6, 2024-01-14:
The combination of a tapered extrude with a 3d offset fails in such a away that it seems impossible to create a tapered cup that isn't upside down:

```python
cup_depth=50.0
cup_radius=35.0
cup_taper=-3
cup_wall=2

with BuildPart() as cup:
    with BuildSketch():
        Circle(cup_radius)
    extrude(amount=cup_depth, taper=cup_taper)
    topf = cup.faces().sort_by(Axis.Z)[-1]
    offset(amount=-cup_wall, openings=topf)
```

```python-traceback
  File "/home/philipsd6/devel/my-3d-models/cup.py", line 80, in <module>
  File "/home/philipsd6/.local/pipx/venvs/cadquery/lib/python3.10/site-packages/build123d/operations_generic.py", line 636, in offset
    solid.offset_3d(openings_in_this_solid, amount, kind=kind).fix()
  File "/home/philipsd6/.local/pipx/venvs/cadquery/lib/python3.10/site-packages/build123d/topology.py", line 1289, in offset_3d
    offset_occt_solid = offset_builder.Shape()
OCP.StdFail.StdFail_NotDone: BRep_API: command not done
```

If `cup_taper` is positive, this works, however that's not the cup shape I want; the solid bottom should be smaller than the open top. If I start building from the top down, using a negative `cup_depth` with a positive `cup_taper`, I get the same error when trying to do the 3d offset.

The only way I've found to get close is to extrude with a positive `cup_depth` and a positive `cup_taper` and then offset with the *bottom* face open. But then the cup is upside down, which adds a semantically unnecessary extra step to reposition.

gumyr, 2024-01-15:
The problem here is with OCCT's handling of the `offset` with `kind=Kind.ARC` which is the default.  If one does `offset(amount=-cup_wall, openings=topf, kind=Kind.INTERSECTION)` the following is generated:
[image]

If the bottom inside edge needs to be filleted, if can be done by adding this:
```py
    inside_edge = cup.edges().sort_by(Axis.Z)[1]
    fillet(inside_edge, cup_wall)
```
[image]

I wish OCCT was better at creating these offsets, but there is nothing that I know of that can be done to improve the situation.

philipsd6, 2024-01-18:
OK, as long as I can avoid the issue by using `kind=Kind.INTERSECTION` that's fine with me. But&hellip; so people like me don't bother you about this, maybe it can automatically fall back to `kind=Kind.INTERSECTION` on error, with a warning?  ¯\\_(ツ)_/¯

gumyr, 2024-01-27:
Instead of automatically trying to fix stuff, I've caught the exception and raised another suggesting and an alternative `kind` might resolve the problem.  This is what the code above does now:
```
Traceback (most recent call last):
  File "/home/roger-maitland/Documents/build123d/src/build123d/topology.py", line 1290, in offset_3d
    offset_occt_solid = offset_builder.Shape()
OCP.StdFail.StdFail_NotDone: BRep_API: command not done

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/roger-maitland/Documents/build123d/experiments/issue-480.py", line 15, in <module>
    offset(amount=-cup_wall, openings=topf)
  File "/home/roger-maitland/Documents/build123d/src/build123d/operations_generic.py", line 636, in offset
    solid.offset_3d(openings_in_this_solid, amount, kind=kind).fix()
  File "/home/roger-maitland/Documents/build123d/src/build123d/topology.py", line 1292, in offset_3d
    raise RuntimeError(
RuntimeError: offset Error, an alternative kind may resolve this error
```
Hopefully that is an acceptable compromise.

closed 2024-01-27 by gumyr