gumyr/build123d #482: "1D" sweeps use wrong OCCT function

MatthiasJ1, 2024-01-15:
The `BRepOffsetAPI_MakePipe` methods used require a G1 spine which is not being enforced by b3d. There is a different constructor which provides fallback parameters for G0 which should be used instead.
That would make it consistent with the other b3d sweep operations

gumyr, 2024-01-16:
I assume you are referring to `Face.sweep`.  I've modified it as:
```py
    @classmethod
    def sweep(cls, profile: Edge, path: Union[Edge, Wire]) -> Face:
        """Sweep a 1D profile along a 1D path"""
        if isinstance(path, Edge):
            path = Wire.make_wire([path])
        # Ensure the edges in the path are ordered correctly
        path = Wire.make_wire(path.order_edges())
        # pipe_sweep = BRepOffsetAPI_MakePipe(path.wrapped, profile.wrapped)
        pipe_sweep = BRepOffsetAPI_MakePipe(
            path.wrapped,
            profile.wrapped,
            # GeomFill_Trihedron.GeomFill_IsConstantNormal,
            # GeomFill_Trihedron.GeomFill_IsCorrectedFrenet,
            # GeomFill_Trihedron.GeomFill_IsDarboux,
            # GeomFill_Trihedron.GeomFill_IsDiscreteTrihedron,
            # GeomFill_Trihedron.GeomFill_IsFixed,
            GeomFill_Trihedron.GeomFill_IsFrenet,
            # GeomFill_Trihedron.GeomFill_IsGuideAC,
            # GeomFill_Trihedron.GeomFill_IsGuideACWithContact,
            # GeomFill_Trihedron.GeomFill_IsGuidePlan,
            # GeomFill_Trihedron.GeomFill_IsGuidePlanWithContact,
            True,
        )
        pipe_sweep.Build()
        return Face(pipe_sweep.Shape())
```
Here I've tried all of the options for how to sweep a polyline as follows:
```py
line_as_wire = Polyline((0, 0), (10, 10), (20, 10)).wire()
trace_pen = line_as_wire.perpendicular_line(1, 0)
traced = Face.sweep(trace_pen, line_as_wire)
```
However, all of the results are as follows:
[image]

MatthiasJ1, 2024-01-17:
```py
spine = Polyline((0,0), (1,10), (10,10))
sect = spine.wire().perpendicular_line(2, 0)
builder = BRepOffsetAPI_MakePipeShell(spine.to_wire().wrapped)
builder.Add(Wire.make_wire([sect]).wrapped, False, False)
builder.SetTransitionMode(BRepBuilderAPI_RightCorner)
builder.Build()
swept = Shape.cast(builder.Shape())
```

[image]

closed 2024-02-04 by gumyr