gumyr/build123d #483: Inconsistent location context inheritance between builders

MatthiasJ1, 2024-01-15:
BuildPart and BuildSketch do not inherit location contexts while BuidLine does *kindof*. How BuildLine inherits context, however, depends on how it is used and can lead to unexpected behavior. How the location of the created object is passed to the parent context is also inconsistent.
These issues are demonstrated below.

[image]

```py
class Collect(Builder):
    objs = []
    def __init__(self, *workplanes):
        super().__init__(*workplanes)
    def _add_to_pending(self, *args, **kwargs): pass
    def _obj(self): return None
    def _add_to_context(self, *objects, **kwargs):
        Collect.objs += [Pos(len(Collect.objs)*3) * o for o in objects]

with Collect():
    with BuildPart(Pos(0,0,10)):
        Sphere(1)

with Collect():
    with BuildPart(Pos(0,0,10)):
        with BuildPart():
            Sphere(1)

with Collect():
    with BuildPart(Pos(0,0,10)):
        with BuildSketch():
            Circle(1)
        extrude(amount=1)

with Collect():
    with BuildSketch(Pos(0,0,10)):
        Circle(1)

with Collect():
    with BuildPart():
        with BuildSketch(Pos(0,0,10)):
            Circle(1)
        extrude(amount=1)

with Collect():
    with BuildPart():
        with BuildSketch(Pos(0,0,10)):
            with BuildLine():
                CenterArc((0,0,0), 1, 0, 360)
            make_face()
        extrude(amount=1)

with Collect():
    with BuildPart(Pos(0,0,10)):
        with BuildSketch(Pos(0,0,10)):
            Circle(1)
        extrude(amount=1)

with Collect():
    with BuildPart():
        with BuildSketch(Pos(0,0,10)):
            with BuildLine(Pos(0,0,10)):
                CenterArc((0,0,0), 1, 0, 360)
            make_face()
        extrude(amount=1)

with Collect():
    with BuildPart():
        with BuildSketch(Plane.XZ.move(Pos(0,0,10))) as s:
            with BuildLine(Plane.XZ) as l:
                Line((-1,0), (1,0))
            make_face(offset(l.line, amount=1))
        extrude(amount=1)

with Collect():
    with BuildPart():
        with BuildSketch(Plane.XZ.move(Pos(0,0,10))) as s:
            with BuildLine(Plane.YZ) as l:
                Line((-1,0), (1,0))
            make_face(offset(l.line, amount=1))
        extrude(amount=1)

with Collect():
    with BuildPart():
        with BuildSketch(Plane.XZ.move(Pos(0,0,10))):
            with BuildLine(Pos(0,0,10)):
                CenterArc((0,0,0), 1, 0, 360)
            make_face()
        extrude(amount=1)

with Collect():
    with BuildPart():
        with BuildSketch(Plane.XZ.move(Pos(0,0,10))) as s:
            with BuildLine(Pos(0,10,0)) as l:
                Line((-1,0), (1,0))
            make_face(offset(l.line, amount=1))
        extrude(amount=1)

with Collect():
    with BuildPart():
        with BuildSketch(Plane.XZ.move(Pos(0,0,10))) as s:
            with BuildLine(Pos(0,0,10)) as l:
                Line((-1,0), (1,0))
            make_face(offset(l.line, amount=1))
        extrude(amount=1)

with Collect():
    with BuildPart():
        with BuildSketch(Plane.XZ.move(Pos(0,0,10))) as s:
            with BuildLine(Plane.XZ.offset(10)) as l:
                Line((-1,0), (1,0))
            make_face(offset(l.line, amount=1))
        extrude(amount=1)

show(Collect.objs)
```

gumyr, 2024-01-16:
Does https://build123d.readthedocs.io/en/latest/build_sketch.html#sketching-on-other-planes help explain what is going on?

When used within the scope of BuildSketch, BuildLine should almost always be working on Plane.XY as **all** sketching is done on Plane.XY and only placed on the provided planes when complete.  Would generating an exception when the user provides a workplane other than Plane.XY to BuildLine while in the scope of BuildSketch be helpful?

MatthiasJ1, 2024-01-18:
https://build123d.readthedocs.io/en/latest/build_sketch.html#sketching-on-other-planes doesn't explain how/why each type of builder behaves differently. It is a simple concept that could be applied consistently to all builders and shouldn't require an explanation.

Also, another note that BuildLine only inherits is parent's location context if it is inside of a BuildSketch and doesn't inside of a BuildPart. This distinction is unnecessary and confusing.

jdegenstein, 2024-01-18:
> Also, another note that BuildLine only inherits is parent's location context if it is inside of a BuildSketch and doesn't inside of a BuildPart. This distinction is unnecessary and confusing.

I wonder if this is related to https://github.com/gumyr/build123d/issues/308

MatthiasJ1, 2024-01-19:
Another odd behavior due to all the moving and case-specific handling of workplanes:
```py
with BuildSketch(Plane.YZ) as s:
    with BuildLine() as l:
        Polyline((0,0), (1,0), (1,1), (0,1), close=True)
    Circle(0.5)
```
[image]

```py
with BuildSketch(Plane.YZ) as s:
    with BuildLine() as l:
        Polyline((0,0), (1,0), (1,1), (0,1), close=True)
    Circle(0.5)
    make_face()
```
[image]

This makes no sense. (I don't mean "it makes no sense to me", I understand what build123d is doing). All these arbitrary special cases and rules could be replaced with a simple universal workplane inheritance hierarchy.

gumyr, 2024-01-20:
This has come up many times in the past, hopefully the following additions to the FAQ explains the reasoning beyond the current design choices: https://build123d.readthedocs.io/en/latest/tips.html#why-doesn-t-buildsketch-plane-xz-work

closed 2024-01-26 by gumyr