gumyr/build123d #486: ValueError("polyline requires three or more pts")

MatthiasJ1, 2024-01-18:
Seems like an unnecessary restriction.

jdegenstein, 2024-01-18:
I agree. I think this should be changed.

gumyr, 2024-01-19:
`Polyline` with two points is just a `Line` which probably means that someone has made a mistake which is why the check is there. Why do you see it as "unnecessary"?

jdegenstein, 2024-01-19:
I see it as unnecessary because `Polyline` could be used in e.g. a programmatic setting (such as a python function) where the number of points is variable. Having to switch between `Line` and `Polyline` based on the number of points when the latter could generate a perfectly valid output seems like an arbitrary choice.

We haven't mentioned `FilletPolyline` which will fillet all "middle" points. Should `FilletPolyline` also change to a minimum of 2 points? I don't have a strong opinion either way.

MatthiasJ1, 2024-01-20:
@jdegenstein exactly. Also from a purely from a semantics point of view, a 2 point polyline is valid. It's the same for `FilletPolyline`.

jdegenstein, 2024-01-31:
Incorporated suggestions into the merged PR here https://github.com/gumyr/build123d/pull/516

closed 2024-01-31 by jdegenstein