gumyr/build123d #517: Line vertex intersection missed if tangential

MatthiasJ1, 2024-02-01:
Both examples intersect at 0,0

---

Returns intersection
```py
a = Line(-1,0)
b = Line((0,0),(0,1))
isect = a.intersect(b)
```
[image]

---

Does not return intersection
```py
a = Line(-1,0)
b = Line((0,0),(1,0))
isect = a.intersect(b)
```
[image]

gumyr, 2024-02-01:
Part of #328

gumyr, 2025-01-15:
`Edge.intersect` has been improved with cd69e6ef022a7bce77ace7d9b5227fab56c142d1.
```py
a1 = Line(-1, 0).edge()
b1 = Line((0, 0), (0, 1)).edge()
isect1 = a1.intersect(b1)
print(f"{isect1=}") # isect1=Vertex(0.0, 0.0, 0.0)

a = Line(-1, 0)
b = Line((0, 0), (1, 0))
isect = a.intersect(b)
print(f"{isect=}") # isect=Vertex(0.0, 0.0, 0.0)
```
The `Edge.intersect` algorithm looks for crossing edges, common vertices and common edges - now with support for `Plane`.  As far as I know this is now complete.