gumyr/build123d #518: Anything.intersect(*B) returns B[0].intersect(*B[1:])

MatthiasJ1, 2024-02-01:
Although this functionality is useful and should be kept in something like `find_all_intersections(List[Shape])`

```py
r = Rectangle(2,2).edges()
l = Line(2,3).edge()
i = l.intersect(*r)
```
[image]

gumyr, 2024-02-01:
Related to Issue #328

gumyr, 2025-01-14:
Some of the intersect code was re-written in the big split.  Currently:

r = Rectangle(2, 2).edges()
l = Line(2, 3).edge()
i = l.intersect(*r)  # None

None is returned from this intersection.  

However, I'll leave this Issue open for now until all of the intersection work is complete to ensure all the cases here are covered.

gumyr, 2025-01-15:
Changed my mind - the `Edge.intersect` work is complete so I'm closing this issue.