gumyr/build123d #520: Circle SVG imported as multiple ellipses

MatthiasJ1, 2024-02-01:
```py
circle_svg = StringIO("""
<svg viewBox="0 0 100 100" xmlns="http://www.w3.org/2000/svg">
  <circle cx="50" cy="50" r="50" />
</svg>
""")

circle = import_svg(circle_svg)
v = circle.vertices()
print(circle.edges()[0].geom_type())
```
```
circle.edges()[0].geom_type() = 'ELLIPSE'
```
[image]

gumyr, 2024-02-01:
build123d uses the svg importer from here: https://github.com/snoyer/ocpsvg.  Try raising the issue there.

snoyer, 2024-10-11:
The underlying issue has been handled in `ocpsvg` and it looks like it fixed this issue in `build123d`
```py
print([e.geom_type for e in import_svg(StringIO('<svg><circle r="1"/></svg>')).edges()])
```
```
[<GeomType.CIRCLE>]
```
maybe @MatthiasJ1 can confirm and close if applicable

gumyr, 2024-10-11:
I get `GeomType.CIRCLE` with ocpsvg 0.2.1. Closing the issue.  Thanks @snoyer!

closed 2024-10-11 by gumyr