gumyr/build123d #530: Shell.make_shell returns Face

MatthiasJ1, 2024-02-03:
```py
section = Rot(90)*CenterArc((0,0), 5, 90, 180)
path = Circle(5).wire()
shell = Shell.make_shell(sweep(section, path))
print(shell.shape_type())
```

```
Face
```

gumyr, 2024-02-03:
`Shell` creation uses the `BRepBuilderAPI_Sewing` class/`SewedShape` method the which has a complex return type of: a `Face`, a `Shell`, a `Solid` or a `Compound`. In this case `make_shell` (which is deprecated, use `Shell(sweep(section,path))`) is being asked to make a `Shell` out of a closed `Compound/Face` the result which can only be a single `Face`.

The `Shell` class could throw an error in this situation but that wouldn't be very helpful.  Do you have a preferred behaviour?

MatthiasJ1, 2024-02-03:
make_shell was deprecated a couple hours before this issue and Shell() has the same behavior. The issue is Solid(Shell()) is broken here #531

gumyr, 2024-02-04:
There is nothing that can be done, this is an OCCT limitation.

closed 2024-02-04 by gumyr