gumyr/build123d #531: Solid from single face does not work

MatthiasJ1, 2024-02-03:
This should produce a sphere
```py
section = Rot(90)*CenterArc((0,0), 5, 90, 180)
path = Circle(5).wire()
shell = Shell(sweep(section, path))
solid = Solid(shell)
```
```
    return ShapeFix_Solid().SolidFromShell(shell.wrapped)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: SolidFromShell(): incompatible function arguments. The following argument types are supported:
    1. (self: OCP.ShapeFix.ShapeFix_Solid, shell: OCP.TopoDS.TopoDS_Shell) -> OCP.TopoDS.TopoDS_Solid

Invoked with: <OCP.ShapeFix.ShapeFix_Solid object at 0x13d194d30>, <OCP.TopoDS.TopoDS_Face object at 0x13d1451f0>
```

gumyr, 2024-02-04:
This is an OCCT limitation - as described before, `Shell` when given a single `Face` returns the `Face` not a `Shell`.  A sphere is the only object that has this limitation, so I guess with OCCT it is not possible to create a `Solid` this way.  There is nothing more that can be done.

MatthiasJ1, 2024-02-04:
Not an OCCT limitation and sphere is not the only shape:
```py
section = Rot(90)*Spline((0,-5), (-3,-2), (-2, 0), (-3, 2), (0,5))
face = sweep(section, Circle(5).wire())
builder = BRepBuilderAPI_MakeShell(BRepAdaptor_Surface(face.face().wrapped).Surface().Surface())
solid = Solid(Shell(builder.Shape()))
```
[image]

Also, revolve is broken here which is why I need to sweep a circle

gumyr, 2024-02-06:
Thanks for pointing this out - I wasn't creative enough to imagine this scenario.  I've updated Shell to allow it to accept either a single Face or a list of Faces with a single member.

```py
x_section = Rot(90) * Spline((0, -5), (-3, -2), (-2, 0), (-3, 2), (0, 5))
surface = sweep(x_section, Circle(5).wire())
single_face = Shell(surface.face())
show(single_face)
```
[image]

closed 2024-02-06 by gumyr