gumyr/build123d #532: Edge.reversed breaks OCCT Topology

MatthiasJ1, 2024-02-03:
"The e.reversed() method is creating new edges with new vertices all the the direction "fixed" which is breaking the connection back to the original vertices." (https://github.com/gumyr/build123d/issues/474#issuecomment-1913639183)

This is supposed to be done with `TopoDS_Shape::Orientation`

gumyr, 2024-02-04:
IIRC `TopoDS_Shape::Orientation` does not actually reverse the edge and doesn't really fix the problem.  Unless you can show how it can be done, this issue will be closed as there is nothing more that can be done.

MatthiasJ1, 2024-02-04:
> IIRC TopoDS_Shape::Orientation does not actually reverse the edge

It do

gumyr, 2024-02-04:
Please show me a working example of `position_at` behaving differently before and after.

MatthiasJ1, 2024-02-05:
https://dev.opencascade.org/doc/overview/html/occt_user_guides__modeling_data.html#occt_modat_5_2_2

gumyr, 2024-02-05:
I've spent a lot of time on time on trying to get this to work in the past and can't afford to spend time on it now.  Maybe in the future I'll look at it again but no guarantees.

baverman, 2024-10-08:
@gumyr I've faced with a related issue when trim isn't working correctly with reversed curves. Currently I try to get into OCCT mess with reversed curves consisting of reversed edges. Hope I could find a solution and make a PR.