gumyr/build123d #533: revolve broken

MatthiasJ1, 2024-02-04:
```py
section = Rot(90)*CenterArc((0,0), 5, 90, 180)
face = revolve(section)
```
Result
```
   upgrader = ShapeUpgrade_UnifySameDomain(self.wrapped, True, True, True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. OCP.ShapeUpgrade.ShapeUpgrade_UnifySameDomain()
    2. OCP.ShapeUpgrade.ShapeUpgrade_UnifySameDomain(aShape: OCP.TopoDS.TopoDS_Shape, UnifyEdges: bool = True, UnifyFaces: bool = True, ConcatBSplines: bool = False)

Invoked with: None, True, True, True
```
Expected:
Sphere surface

https://github.com/gumyr/build123d/issues/531#issuecomment-1925548022

gumyr, 2024-02-04:
`revolve` doesn't take `Edges` or `Wires` just `Faces`:
```py
def revolve(
    profiles: Union[Face, Iterable[Face]] = None,
    axis: Axis = Axis.Z,
    revolution_arc: float = 360.0,
    clean: bool = True,
    mode: Mode = Mode.ADD,
) -> Part:
```
If you do:
```py
profile = split(Rot(90) * Circle(5), Plane.YZ)
revolved = revolve(profile)
```
you get:
[image]

I'm guessing you would like it to be extended to create `Faces` from 1D shapes?

MatthiasJ1, 2024-02-05:
Yes, that is the logical progression. I updated the title.