gumyr/build123d #534: Bezier argument auto list parsing incorrect

MatthiasJ1, 2024-02-04:
```py
a = Bezier((0,-5), (0,5))
b = Bezier([0,-5], [0,5])
```

[image]

gumyr, 2024-02-04:
This is fundamentally an issue with how accepting `Vector` is, i.e. this is valid:
```
>>> Vector(1)
Vector: (1.0, 0.0, 0.0)
```
If we look at the debugger while processing: `Bezier([0,-5], [0,5])` we see this:
[image]
`cntl_pnts` is the raw inputs flattened, here the two lists are flattened into four points: `[0, -5, 0, 5]` which gets turned into four `polls`/`Vectors` (three of which are unique): `Vector(0,0,0), Vector(-5,0,0), Vector(0,0,0), Vector(5,0,0)`.  This ultimately results in a horizontal line.

So, is `Vector` too accepting? It's possible to do this currently `Line(-5,5)` which would create the same horizontal line as `b`.  

Should `Bezier` (and other objects) reject multiple lists?  Should inputs like `Bezier((0,0), other_points)` be rejected?

At this point I'm leaning towards saying that points must be tuples, so `(5,)` would be acceptable, but not `5`. Therefore, `[0,-5], [0,5]` would be rejected as containing no points.

Thoughts?

MatthiasJ1, 2024-02-05:
The list expansion only makes sense in the case `len(vargs)==1` and adding that condition will fix these issues.

gumyr, 2026-10-08:
The common cause is `_is_point` in `build_common.py`, which only recognises a *tuple* of numbers as a point, so `flatten_sequence` recurses into a list of numbers.