gumyr/build123d #536: Locations should reference a single underlying shape

MatthiasJ1, 2024-02-04:
Currently it duplicates the underlying geometry. For a simple usecase like `GridLocations`, OCCT provides helpers like `BOPAlgo_MakePeriodic`.

gumyr, 2024-02-04:
If `Locations` were to deep copy objects such that they share the same underlying TShape then changes to one would impact them all which may not be what the user wants.  One could use `Locations` to place shallow copies if that is desired.

MatthiasJ1, 2024-02-05:
> If Locations were to deep copy objects such that they share the same underlying TShape
> One could use Locations to place shallow copies if that is desired.

Deep copying the objects would mean that they do not share the same underlying `TShape`. A shallow copy would share the same `TShape`.
`Locations` currently deep-copies objects:
```py
boxes = Compound(GridLocations(2,0,2,1) * Box(1,1,1))
print(boxes.show_topology("Solid"))
print(f"{boxes.solids()[0].wrapped.IsPartner(boxes.solids()[1].wrapped)}")
```
```
Compound      at 0x100ab30f0, Center(0.0, 0.0, 0.0)
├── Compound  at 0x1413a17f0, Center(-1.0, 0.0, 0.0)
│   └── Solid at 0x1504cb430, Center(-1.0, 0.0, 0.0)
└── Compound  at 0x100ab3db0, Center(1.0, 0.0, 0.0)
    └── Solid at 0x1504c9b30, Center(1.0, 0.0, 0.0)

False
```

---

> changes to one would impact them all which may not be what the user wants

Practically every B3D operation creates a new object rather than modifying the original which means that performing operations on a shallow copy would not the copy itself and therefore would not impact the other location copies.

closed 2024-02-05 by gumyr