gumyr/build123d #540: Inconsistent builder to geometry promotion

MatthiasJ1, 2024-02-06:
```py
with BuildSketch() as s: Circle(10)

with BuildSketch(XY.offset(10)) as s2: add(s) # Ok

with BuildLine() as l: Line((0,0,0), (0,0,1))

def test(f):
    def _test(f):
        try: eval(f)
        except Exception as e: return type(e).__name__
        return True
    print(f"{f}:", "ok" if ((e:=_test(f))==True) else f"unsupported ({e})")

test("extrude(s, 1)")
test("revolve(s, Y.located(Pos(30)))")
test("sweep(s, l)")
test("offset(s, 1)")
test("mirror(s)")
test("split(s)")
test("scale(s)")
test("s+s")
test("trace(l)")
test("bounding_box(s)")

test("loft([s, s2])")
test("loft([s, s2.sketch])")
test("loft([s.sketch, s2])")
test("loft([s.sketch, s2])")

test("Compound(s)")
test("Compound([s])")
```
```
extrude(s, 1): ok
revolve(s, Y.located(Pos(30))): ok
sweep(s, l): ok

offset(s, 1): unsupported (TypeError)
mirror(s): unsupported (TypeError)
split(s): unsupported (AttributeError)
scale(s): unsupported (AttributeError)
s+s: unsupported (RuntimeError)
trace(l): unsupported (TypeError)
bounding_box(s): unsupported (AttributeError)

loft([s, s2]): unsupported (Standard_Failure)
loft([s, s2.sketch]): ok
loft([s.sketch, s2]): unsupported (Standard_Failure)
loft([s.sketch, s2]): unsupported (Standard_Failure)

Compound(s): ok
Compound([s]): unsupported (AttributeError)
```

gumyr, 2024-02-06:
All of these errors are caused by passing a Builder where a Shape is expected.

As mentioned in issue #538, it's up to the user to pass the correct type arguments. It isn't worth the effort to attempt to fix the user's mistakes throughout the whole API.

closed 2024-02-06 by gumyr

MatthiasJ1, 2024-02-07:
My point is that a bunch of operations accept a Builder instead of a Shape while others don't and this behavior is inconsistent. Some operations like loft support Builders as the first argument but not the rest.