gumyr/build123d #541: Inconsistent errors for unsupported argument type

MatthiasJ1, 2024-02-06:
```py
with BuildSketch() as s: Circle(10)
with BuildSketch(XY.offset(10)) as s2: add(s) # Ok
with BuildLine() as l: Line((0,0,0), (0,0,1))
```
```
offset(s, 1): unsupported (TypeError)
mirror(s): unsupported (TypeError)
split(s): unsupported (AttributeError)
scale(s): unsupported (AttributeError)
s+s: unsupported (RuntimeError)
trace(l): unsupported (TypeError)
bounding_box(s): unsupported (AttributeError)
loft([s, s2]): unsupported (Standard_Failure)
loft([s.sketch, s2]): unsupported (Standard_Failure)
loft([s.sketch, s2]): unsupported (Standard_Failure)
Compound([s]): unsupported (AttributeError)
```

gumyr, 2024-02-06:
All of these errors are caused by passing a Builder where a Shape is expected.

As mentioned in issue https://github.com/gumyr/build123d/issues/538, it's up to the user to pass the correct type arguments. It isn't worth the effort to attempt to fix the user's mistakes throughout the whole API.

closed 2024-02-06 by gumyr

MatthiasJ1, 2024-02-07:
In this issue I am not saying anything about whether these functions should accept Builders, just that they should return a consistent error when an unsupported argument is passed in. Some of them have no indication that the error is caused by the input being a builder which is confusing.