gumyr/build123d #546: Add `Plane.reverse()` method

tobywf, 2024-02-09:
It might be good to add a `Plane.reverse()` method, analogue to [`Axis.reverse()`](https://github.com/gumyr/build123d/blob/4641d1d8dbb7db394242b0297c9b83e5bafb89e1/src/build123d/geometry.py#L672) and [`Vector.reverse()`](https://github.com/gumyr/build123d/blob/4641d1d8dbb7db394242b0297c9b83e5bafb89e1/src/build123d/geometry.py#L306).

Pros:
* `Axis` implements both [`__neg__`](https://github.com/gumyr/build123d/blob/4641d1d8dbb7db394242b0297c9b83e5bafb89e1/src/build123d/geometry.py#L676) and [`reverse()`](https://github.com/gumyr/build123d/blob/4641d1d8dbb7db394242b0297c9b83e5bafb89e1/src/build123d/geometry.py#L672), `Vector` implements both [`__neg__`](https://github.com/gumyr/build123d/blob/4641d1d8dbb7db394242b0297c9b83e5bafb89e1/src/build123d/geometry.py#L385) and [`reverse()`](https://github.com/gumyr/build123d/blob/4641d1d8dbb7db394242b0297c9b83e5bafb89e1/src/build123d/geometry.py#L306), while `Plane` currently only implements [`__neg__`](https://github.com/gumyr/build123d/blob/4641d1d8dbb7db394242b0297c9b83e5bafb89e1/src/build123d/geometry.py#L1918).
* I spent longer than I care to admit trying to figure out how to reverse/invert a plane. `Plane.reverse()` helps with discoverability due to auto-complete.
* A method has a clear operator precedence. `-Plane.XY.offset(10.0)` applies the offset first, and then the reversal. `(-Plane.XY).offset(10.0)` is required to first reverse and then offset.`Plane.XZ.reverse().offset(10.0)` works as intended.

Cons:
* `Axis.reverse()` corresponds directly to the OCCT method `Reversed`. `Plane.reverse()` doesn't; but `Vector.reverse()` also doesn't.
* There would be two ways to reverse a plane. But that's true of `Axis` and `Vector` also.
* There's already `Plane.reverse_transform()`, which is close in name.
* Could be that e.g. `invert()` is a more appropriate name.

Overall, I think `Plane.reverse()` makes sense, but let me know. Looks like an easy PR; happy to submit one if you'd like.

tobywf, 2024-02-09:
Example of operator precedence issue:

```py
Plane.reverse = lambda s: -s
print(-Plane.XY.offset(10.0))
print((-Plane.XY).offset(10.0))
print(Plane.XY.reverse().offset(10.0))
```

```
Plane(o=(0.00, 0.00, 10.00), x=(1.00, 0.00, 0.00), z=(-0.00, -0.00, -1.00))
Plane(o=(0.00, 0.00, -10.00), x=(1.00, 0.00, 0.00), z=(-0.00, -0.00, -1.00))
Plane(o=(0.00, 0.00, -10.00), x=(1.00, 0.00, 0.00), z=(-0.00, -0.00, -1.00))
```

gumyr, 2024-02-10:
Done.

closed 2024-02-10 by gumyr