gumyr/build123d #547: Add `2d_approx` option for 2d offsets

jdegenstein, 2024-02-10:
Propose to add a `2d_approx: bool = False` default to `operationgs_generic.offset` that will ultimately get passed down to `Mixin1D.offset_2d`

This option will provide more robust offsetting behavior at the expense of potentially adding many more vertices. The OCCT 7.7.0 `SetApprox` option only returns circular arcs and straight lines (instead of the many other possible geom types).

Better behavior from OCCT or other workarounds at the build123d level are the better long term solution. This provides an easy to apply workaround with minimal effort.

gumyr, 2024-02-10:
Sounds good to me.  Ideally this would be fixed in OCCT so having a good example of how this will fail in 2D but pass in 3D would be very valuable.

gumyr, 2024-10-24:
from @jdegenstein:
 Further investigation: if SetApprox is True and an offset face were extruded, the resulting shape will have a lot more edges potentially because a spline broken up into hundreds of small circles arcs. So what, you might ask? The next problem is that OCCT doesn't like when edges are consumed by e.g. fillets/chamfers meaning that the size of fillet/chamfer you could apply would be limited.

This is a really annoying problem, it would be best for OCCT to just fix the algo, which they are obviously capable of doing since 3D offsets of the same shapes work just fine.

from @jdegenstein:
Is this worth it for 2D offsets that are more reliable? My vote is that SetApprox should be False by default, and maybe automatically roll over to True if the False attempt fails? EDIT: and MAYBE warn the user 

From #745 (@Spectre5),
```py
# base part shell
shell_thickness = 1 / 16 * IN
sketch = Sketch() + [
    Pos(Z=0.00 * IN) * Rectangle(6.0 * IN, 4.0 * IN),
    Pos(Z=1.00 * IN) * Rectangle(5.0 * IN, 3.0 * IN),
]
solid = loft(sketch, ruled=True)
bottom = solid.faces().sort_by()[0]
shell = offset(solid, amount=-shell_thickness, openings=bottom)

# create a a boss on the inside surface
face = shell.faces().sort_by()[-2]
# face = shell.faces().sort_by()[-1]
outer = offset(face, amount=-0.25 * IN).face()
```
`outer` appears to be a valid Face but silently creates a Solid that can't be fused to the rest of the part when extruded.

There seems to be potentially three things that could be done here:
1) Expose the `SetApprox` parameter and let the user figure it out.
2) Detect when the result is invalid and try again with `SetApprox(True)` while issuing a warning to the user.
3) Thicken the Face and use 3D offset with the opening being the original Face then return the offset from the 3D object.

Comments about these options:
1) The least amount of work but also the least helpful as the user needs to know a lot to make this useful.
2) I don't know how to detect that the Face is bad - any ideas?
3) My gut feeling is that there will be a whole bunch of new problems but I hope I'm wrong.

gumyr, 2024-10-24:
FreeCAD has a  [BRepOffsetAPI__MakeOffsetFix](https://freecad.github.io/SourceDoc/d8/dff/classPart_1_1BRepOffsetAPI__MakeOffsetFix.html) class, @jdegenstein with the work that is being done with OCCT, is it possible to use code from FreeCAD as well?

Spectre5, 2024-10-24:
Obviously the ideal solution would be for OCCT to fix this, and I think they should.  Do you know if there is a bug reported for this already?  I imagine there must be, and we should add a comment or additional info to it.  Otherwise create one.

As for the options above, I think 1 makes sense to implement.  Keep the default as False and then I see it mainly as a documentation effort.  Just make sure it is clear in the doc string/documentation what exactly it does and what the ramifications and potential issues are that it may cause.

I think for item 2, it could easily be added in the future if a good way to detect the bad face is found.  Then if detected, it could raise an exception or put a warning to the user to indicate that a bad face was detected and they may want to consider setting approximate to True (i.e. just tell them, but don't do it automatically).

As for the third item, can you give a demonstration of how this would work?  Maybe edit my example on the other issue to show it.  I'm not familiar enough with this to have any opinion on it, but it seems interesting if it would be more robust, though a bit more convoluted.

Are yous sure that BRepOffsetAPI_MakeOffsetFix is meant to fix this same issue?  Or is it only meant to fix the issue of having a single wire, which build123d already addresses in topology.py

```python
        # Avoiding a bug when the wire contains a single Edge
        if len(line.edges()) == 1:
            edge = line.edges()[0]
            edges = [edge.trim(0.0, 0.5), edge.trim(0.5, 1.0)]
            topods_wire = Wire(edges).wrapped
        else:
            topods_wire = line.wrapped
```

gumyr, 2024-10-24:
The question about  `BRepOffsetAPI_MakeOffsetFix` was not directly connected to this problem - sorry for the confusion. I'm know some work is being done with OCCT changes and wondered if inclusion of these type of thing was possible.

jdegenstein, 2024-10-25:
I do NOT think that there should be an automatic rollover from non-approx to approx. We will be inundated by problems relating to downstream operations as you mentioned above.

It might be possible to add [BRepOffsetAPI__MakeOffsetFix](https://freecad.github.io/SourceDoc/d8/dff/classPart_1_1BRepOffsetAPI__MakeOffsetFix.html) to https://github.com/jdegenstein/ocp-addons but I have not investigated it at all yet

Spectre5, 2025-03-16:
Has there been anymore consideration to this?  Adding a keyword argument to control `SetApprox `.

gumyr, 2025-03-17:
I haven't had a chance to look into this.