gumyr/build123d #549: split(fillet) broken

MatthiasJ1, 2024-02-14:
```py
b = extrude(Rectangle(10,10), 10)
b = split(b, Plane.XY.offset(2))
```

[image]

```py
b = extrude(Rectangle(10,10), 10)
b = fillet(b.edges()<<Axis.Z, 4)
b = split(b, Plane.XY.offset(2))
```

[image]

bernhard-42, 2024-02-15:
This works:
```python
b = extrude(Rectangle(10,10), 10)
b = fillet(b.edges()<<Axis.Z, 4).solid()  # <===
b = split(b, Plane.XY.offset(2))
```

It seems to be a similar same problem as with `extrude( ..., both=True)`
```python
obj = SlotCenterPoint((0, 0), (-6, 0), 5) - Pos(-6, 0) * Circle(1)
p2 = extrude(obj, 0.5, both=True)
s2 = split(p2, Plane.YZ.offset(-3), keep=Keep.BOTTOM)
``` 
fails, while 
```python
obj = SlotCenterPoint((0, 0), (-6, 0), 5) - Pos(-6, 0) * Circle(1)
p2 = extrude(obj, 0.5, both=True).solid()  # <===
s2 = split(p2, Plane.YZ.offset(-3), keep=Keep.BOTTOM)
```
succeeds.

Somehow in algebra mode a step is missing, because this works:

```python
with BuildPart() as p:
    with BuildSketch() as s:
        Rectangle(10,10)
    extrude(amount=10)
    fillet(p.edges()<<Axis.Z, 4)
    split(bisect_by=Plane.XY.offset(2))
```

and this works:

```python
with BuildPart() as p:
    with BuildSketch() as s:
        SlotCenterPoint((0, 0), (-6, 0), 5)
        with Locations((-6, 0)):
            Circle(1, mode=Mode.SUBTRACT)
    extrude(amount=0.5, both=True)
    split(bisect_by=Plane.YZ.offset(-3), keep=Keep.BOTTOM)
```

I guess this is something that happens in the builder at the end, but I can't find it.
Maybe @gumyr  knows from experience what happens in the Builder that might need to be added to the direct API for algebra mode working without adding `.solid()`

gumyr, 2024-02-15:
The problem here was multiple layers of `Compound` were getting added in some circumstances.  I've changed the `extrude` operation to `return Part(ShapeList(new_solids).solids())` which removes the extra Compound layer (which is what happens in the Builder when it's combined).  I didn't change anything else because sometimes alternative shapes may be okay.

If you see any more of these, please let me know.

closed 2024-02-15 by gumyr