gumyr/build123d #554: How to chamfer extruded text

Paul8043, 2024-02-19:
How to chamfer an extruded text? I want to chamfer all edges of the top-face. This works fine for a simple box, but not for text.
```
from build123d import *
from ocp_vscode import *

set_port(3939)

b   = Pos(0,30,0)*Box(10,10,10)
c1  = Pos(0,15,0)*chamfer(b.edges().group_by(Axis.Z)[-1],1)

t2d = Text("ABCD",font_size=50,font_style=FontStyle.BOLD)
t3d = extrude(t2d,10)

c2  = Pos(0,-30,0)*chamfer(t3d.edges().group_by(Axis.Z)[-1],1)  # trouble

show_all()
```
The line marked with "trouble" generates various errors (see below):

[image]

gumyr, 2024-02-19:
Your code works for me (on Ubuntu 23.04) which almost certainly narrows the problem down to the font you are using. Text is notoriously difficult to work with as what looks look often doesn't work in a CAD system - short or intersecting edges are often the problem. Hopefully by selecting another font (i.e. set the `font=` parameter) you'll find one you're happy with. 
[image]

Paul8043, 2024-02-19:
We use different platforms, I have run he code on Windows 11 Home (see below):
[image]
It could findout which fonts are installed, most of them are not so pleasant for my purpose. If possible, I want to keep the default-font ("ARIAL"). During my further investigations, I have replaced chamfer() by filet(),
```
from build123d import *
from ocp_vscode import *

set_port(3939)
show_clear()

b   = Pos(0,30,0)*Box(10,10,10)
c1  = Pos(0,15,0)*chamfer(b.edges().group_by(Axis.Z)[-1],1)

t2d = Text("ABCD",font="Arial",font_size=24,font_style=FontStyle.BOLD)
t3d = extrude(t2d,2)

#c2  = Pos(0,-30,0)*chamfer(t3d.edges().group_by(Axis.Z)[-1],1)  # bad
c3  = Pos(0,-25,0)*fillet(t3d.edges().group_by(Axis.Z)[-1],1)          # good

show_all()
```
and the code runs fine! I can live with this alternative (rounded edges).

[image]

Due to this experiment, I would conclude that the root-cause is not font related, but somewhere hidden inside the chamfer()-routine.