gumyr/build123d #577: Hole operations affect unrelated geometry above target surface

MatthiasJ1, 2024-02-29:
```py
with BuildPart() as p:
    Sphere(10)
    Box(20,20,5, mode=Mode.SUBTRACT)
    with Locations((faces()|Axis.Z)<<Axis.Z):
        # Hole(3)
        CounterBoreHole(3, 8, 2)
        # CounterSinkHole(3, 8)
```

[image]

MatthiasJ1, 2024-02-29:
Here is another example of attempting to create a hinged box (yes I know you could do this in such a way that would avoid this issue, but the IMO intuitive way of creating the tabs then adding a countersunk hole on the tabs does not work due to this issue)
```py
with BuildPart() as p:
    Box(50, 30, 30)
    with BuildPart():
        with Locations((0,-18,10)):
            Box(50, 10, 10, align=(Align.CENTER, Align.MIN, Align.CENTER))
            Cylinder(5, 50, rotation=(0,90,0))
            Box(20, 30, 30, mode=SUBTRACT)
        with Locations(((faces()|X)<X)[0::2]):
            with Locations((0,3)):
                CounterSinkHole(2, 4)
```
[image]

MatthiasJ1, 2024-02-29:
Or a simpler example in which the workaround would be much more unintuitive.
```py
with BuildPart() as p:
    Box(50, 30, 30)
    with Locations((0,-15,0)):
        Box(20, 30, 30, mode=SUBTRACT)
    with Locations(((faces()|X)<X)[2]):
        CounterSinkHole(3, 6)
```
[image]

gumyr, 2024-03-01:
This behavior is by design to simplify creating through holes and counter-sink/bore holes that are not perpendicular to the surface.  The original behavior was as you describe but changed as a result of this issue: https://github.com/gumyr/build123d/issues/4

Another parameter could be added to the hole objects to control this though.