gumyr/build123d #582: [documentation] Some code snippets in documentation are not rendering nice

mnesarco, 2024-03-02:
In some places, the python code is rendered as a string commet.

i.e. https://build123d.readthedocs.io/en/latest/build_part.html#implicit-parameters

[image]

jdegenstein, 2024-03-02:
Looks like this was a result of the examples being updated. I have fixed this one https://github.com/gumyr/build123d/commit/20044572fac1f89c720537bfb0a7c85527befa85

Let me know if you find others, I will keep this issue open for now assuming there are more to fix.

gumyr, 2024-03-04:
The tea cup is messed up in the same way on the main build123d page.

jdegenstein, 2024-03-04:
This page is also messed up https://build123d.readthedocs.io/en/latest/tutorial_selectors.html -- probably best to switch over to the "start-before ..." approach from my earlier commit for everything from now on.

gumyr, 2024-03-17:
The ability to use both `start-before` and the other lines commands is awesome.  Here is an example from the selector tutorial:
```rst
.. literalinclude:: selector_example.py
    :start-after: [Code]
    :end-before: [End]
    :lines: 1-9,18-24,33-34
    :emphasize-lines: 10-16
```
All of the above problems are fixed.  Please let me know if you find any more.

closed 2024-03-17 by gumyr