gumyr/build123d #583: bad interface of a `Color()` constructor

cyberhuman, 2024-03-03:
I needed to create Color values from a hexadecimal color code, so naturally I looked up the source code and noticed there exists a constructor that takes an integer:
https://github.com/gumyr/build123d/blob/2ae444bad54a00c8ebf344c828ade2e1a6a15c02/src/build123d/geometry.py#L892-L897
For a second I thought that it was almost exactly what I needed but then I noticed the sloppy part:
> 0xRRGGBB **or 0xRRGGBBAA**

This means that it's not possible to use this constructor to create a semi-transparent green or blue color, or an arbitrary color in the whole RGBA range. For example `0x00FF00 80` might be a 50% transparent green, but it's recognized as a pink color #FF0080 instead.

IMO this kind of interface is very bad. It's unreliable and can be used only by humans and only after carefully considering the value that is passed in. And for humans, it's easier to use color names or specify RGB values separately.

Hence I believe this constructor must be either removed altogether, or it must accept the color code in an unambiguous format, for example 0xAARRGGBB. What do you think?

gumyr, 2024-03-04:
That was some silly code - thanks for pointing it out.  I've changed the hex API to:
```py
    def __init__(self, color_code: int, alpha: int = 0xFF):
        """Color from a hexidecimal color code with an optional alpha value

        Args:
            color_code (hexidecimal int): 0xRRGGBB
            alpha (hexidecimal int): 0x00 <= alpha as hex <= 0xFF
        """
```
Please let me know what you think.

cyberhuman, 2024-03-05:
@gumyr yes, that one is much better! Thank you!

closed 2024-03-05 by gumyr