gumyr/build123d #584: Weird behavior with offset and buildline after update to 0.4

Ruudjhuu, 2024-03-05:
I experience some weird behavior regarding the `offset` function in combination with a sketch created with `BuildLine`.

## Problem
(see code snippet below)
|||
|-|-|
| GIVEN  | a sketch is created with  `BuildSketch`, `Buildline`, `Polyline` and `make_face`.|
|WHEN   |the sketch is used in another `BuildPart`  + `BuildSketch` context by using `add`|
|AND      |an `offset` call is executed|
|THEN    |sketch is flipped <-- issue |
|AND | sketch is bigger by offset amount

## Expectations
|||
|-|-|
|THEN |sketch stays in same location|
|AND |  bigger by offset amount.|

## Information
Build123d version: 0.4.0
Happens in earlier version? Not in 0.3.0

Before offset call:
[image]

After offset call:
[image]

When using the commented out `Triangle` call which creates exactly the same sketch the behavior is as expected.

My intention with this construction is a little bit more complex and involves multiple objects, but this is the minimal code snippet I could reproduce the issue with.

This issue currently stops me from using 0.4.0

Code snippet. 
```python
from build123d import *

with BuildSketch() as profile_a:
    # Triangle(a=2, b=2, C=90, align=Align.MIN)
    with BuildLine():
        Polyline(
            (0, 0),
            (2, 2),
            (2, 0),
            close=True,
        )
    make_face()

with BuildPart() as part, BuildSketch(Plane.XZ) as profile, Locations((10, 0)):
    add(profile_a)
    offset(amount=1, kind=Kind.INTERSECTION)

```

jdegenstein, 2024-03-05:
I am able to reproduce the above on build123d release 0.4, but not able to reproduce on a newer version -- 0.4.1.dev39+g659ce3a nor 0.4.1.dev50+g8632e40 (latest commit). I used this slightly modified snippet to avoid empty buildpart warnings:
```py
with BuildSketch() as profile_a:
    # Triangle(a=2, b=2, C=90, align=Align.MIN)
    with BuildLine():
        Polyline(
            (0, 0),
            (2, 2),
            (2, 0),
            close=True,
        )
    make_face()

with BuildPart() as part:
    with BuildSketch(Plane.XZ) as profile:
        with Locations((10, 0)):
            add(profile_a)
            offset(amount=1, kind=Kind.INTERSECTION)
    Box(.1,.1,.1)
```

Ruudjhuu, 2024-03-05:
I can can confirm the issue does not exist on the latest commit. I'll wait for a new release. Thanks for the help.

closed 2024-03-05 by Ruudjhuu