gumyr/build123d #590: Basic operation results in invalid geometry

MatthiasJ1, 2024-03-17:
```py
p = Box(2,2,2) - Pos(-1.5) * Sphere(1)
```
[image]

gumyr, 2024-03-17:
This is an OCCT problem with how the `Edge` of the sphere is aligned.  Consider this version:
```py
s = Pos(-1.5) * Sphere(1).rotate(Axis.Y, 180)
p = Box(2, 2, 2) - s
```
[image]
Note how the `Edge` of the sphere is rotated away from the cube which allows the boolean operation to work as expected.

I've considered constructing a sphere from two hemispheres but an `clean` operation would fuse them together again which doesn't solve the problem.  This really needs to be fixed at the OCCT level.

MatthiasJ1, 2024-03-17:
Another issue from this scenario: This creates valid geometry but the exported STEP file gives the union rather than subtraction.

```py
p = Box(2,2,2) - Pos(0.7) * Sphere(0.7)
```
[image]
[image]

gumyr, 2024-03-17:
I'm guessing the OCCT requires all Solids to have at least one Edge but it breaks so many things.  I expect fixing it in OCCT would be difficult but I don't know.

MatthiasJ1, 2024-03-18:
OCCT fundamentally requires surfaces to be closed in their parametric space. Removing that requirement in OCCT would be a monumental effort, however, the geometries here should be able to comply within that limitation.

MatthiasJ1, 2024-03-18:
I believe it should be possible (it only needs to be closed in the 2d parametric space of the surface and you can create a circle with a single vertex and edge, so I imagine a clipped sphere shell is topologically equivalent) to create a parametrically closed sphere like the one below without any seams on the spherical surface but I have been unsuccessful so far. It would be useful in many scenarios to circumvent the seam issue.

[image]

jdegenstein, 2026-09-15:
FYI I reported the issue to OCCT and have a tested working patch for this issue posted there. https://github.com/Open-Cascade-SAS/OCCT/issues/1541