gumyr/build123d #605: Regression in ExportSVG for Python 3.10

openvmp, 2024-04-02:
PartCAD started throwing an exception for build123d parts at some point last week
(always tries to use the latest version of build123d, for now).
I was able to reproduce the same exception with the following script.
It throws an exception after an update in 3.10 but not in 3.11:

```python
import build123d as b3d
exporter = b3d.ExportSVG()
```

```
$ python test.py
Traceback (most recent call last):
  File "/tmp/1/test.py", line 3, in <module>
    exporter = b3d.ExportSVG()
  File "/home/clairbee/projects/robotics/partcad/.conda/lib/python3.10/site-packages/build123d/exporters.py", line 944, in __init__
    self.add_layer(
  File "/home/clairbee/projects/robotics/partcad/.conda/lib/python3.10/site-packages/build123d/exporters.py", line 991, in add_layer
    layer = ExportSVG._Layer(
  File "/home/clairbee/projects/robotics/partcad/.conda/lib/python3.10/site-packages/build123d/exporters.py", line 907, in __init__
    self.fill_color = convert_color(fill_color)
  File "/home/clairbee/projects/robotics/partcad/.conda/lib/python3.10/site-packages/build123d/exporters.py", line 902, in convert_color
    if isinstance(c, RGB):
  File "/home/clairbee/projects/robotics/partcad/.conda/lib/python3.10/typing.py", line 994, in __instancecheck__
    return self.__subclasscheck__(type(obj))
  File "/home/clairbee/projects/robotics/partcad/.conda/lib/python3.10/typing.py", line 997, in __subclasscheck__
    raise TypeError("Subscripted generics cannot be used with"
TypeError: Subscripted generics cannot be used with class and instance checks
```

```
$ python --version
Python 3.10.13
```

openvmp, 2024-04-02:
Ubuntu 
bookworm/sid
Linux hana 6.1.0-1035-oem #35-Ubuntu SMP PREEMPT_DYNAMIC Wed Feb 21 10:58:23 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

jdegenstein, 2024-04-02:
I am not able to reproduce on python 3.10. Can you provide your installed versions of build123d, ocpsvg, ezdxf? I believe there was a discussion on discord that gumyr recommended upgrading the version of ezdxf. I am not sure what versions of ezdxf are affected but we may need to update the requirements in `pyproject.toml`.

@gumyr by the way, I noticed the notes for apple silicon installs are incomplete since they do not install ezdxf. https://build123d.readthedocs.io/en/latest/installation.html#special-notes-on-apple-silicon-installs

I can fix that but I would like to know more about the version requirements for ezdxf.

gumyr, 2024-04-02:
I noticed the ezdxf problem on my machine but not on the build servers so I thought it might be a version thing.  After upgrading it went away but I unfortunately didn't record the appropriate version numbers.  I have 1.1.4 now so maybe that should be the minimum?

openvmp, 2024-04-03:
ezdxf==1.0.3
ocpsvg==0.2.0
build123d==0.5.0

openvmp, 2024-04-03:
Upgrading ezdxf to 1.1.0 made the problem go away. Thanks!

jdegenstein, 2024-04-03:
Thank you very much for the information, I will fix the two issues mentioned above.

jdegenstein, 2024-06-06:
closed with merged PR https://github.com/gumyr/build123d/pull/638

closed 2024-06-06 by jdegenstein