gumyr/build123d #611: Chamfer broken when selecting by face

MatthiasJ1, 2024-04-24:
This is useful in the scenario when you are trying to select edges on the inner or outer wires of a face of a solid.

---

```py
p = Cylinder(5,10)
p = chamfer((p.faces()>>Axis.Z)[0].edges(), 1)
```
```
ValueError: 2D chamfer operation takes only Vertices
```
---
```py
p = extrude(Circle(5), 10)
p = chamfer((p.faces()>>Axis.Z)[0].edges(), 1)
```
```
ValueError: 2D chamfer operation takes only Vertices
```
---
```py
p = Cylinder(5,10)
p = p.chamfer(1, None, (p.faces()>>Axis.Z)[0].edges())
```
```
TypeError: Cylinder.__init__() missing 1 required positional argument: 'height'
```
---
```py
p = extrude(Circle(5), 10)
p = p.chamfer(1, None, (p.faces()>>Axis.Z)[0].edges())
```
```
This one works.
```

westurner, 2025-06-29:
To see the actual exception with bin/python (because vscode debugger obscures the exception) :
```py
$ python ./model2.py
# ...
  File "./model2.py", line 62, in mushroom
    stem = stem.chamfer(edge_list=(stem.faces()>>Axis.Z)[0].edges(), length=stem_chamfer_length, length2=0)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.12/site-packages/build123d/topology/three_d.py", line 257, in chamfer
    new_shape = self.__class__(chamfer_builder.Shape())
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Cylinder.__init__() missing 1 required positional argument: 'height'
```

The "Extrude a circle and selecting faces from that isntead" chamfer workaround works here too.

Would something like this work?

```py
_shape = chamfer_builder.Shape()
new_shape = self.__class__(*_shape.args, **_shape.kwargs)
```

Or there could be a .copy() method interface for Shapes?

westurner, 2025-06-29:
Maybe `shape2 = shape.copy(**kwargs_to_merge)`