gumyr/build123d #612: Builder mode performance issues

MatthiasJ1, 2024-04-25:
In the following example, the Builder version is 16x slower:

```py
with print_time():
    p = Cylinder(20, 20)
    p += TrapezoidalThread(2*20+2, 2, 60, 10)

with print_time():
    with BuildPart() as p2:
        Cylinder(20, 20)
        TrapezoidalThread(2*20+2, 2, 60, 10)
```
gives:
```
0.5091060638427734s
8.107296800613403s
```

gumyr, 2024-04-26:
This is very interesting - the difference in time is just in time it takes to call `TrapezoidalThread`.  The Algebra result also looks correct while the Builder version is missing some of the thread - curious! Some good clues here.

jdegenstein, 2024-04-26:
Here's another slightly tweaked version, that predictably works more quickly and interestingly is not missing any thread:
```py
from build123d import *
from bd_warehouse.thread import *
import time
start_time = time.time()
th = TrapezoidalThread(2*20+2, 2, 60, 10)
with BuildPart() as p3:
    Cylinder(20, 20)
    add(th)
print("--- %s seconds ---" % (time.time() - start_time))
```
Time is approximately 2x the algebra version above.

gumyr, 2024-04-26:
I think I know what is happening here.  The thread is a `Compound`/assembly composed of 4 complete loops and a top and bottom loop (6 `Solids`).  Builder mode extracts the `Solids` for the fuse operation (while Algebra fuses the `Compound`) which is causing the performance degradation.  Now I have to figure out why I coded it this way :)